grblHAL/core #616: communication with stm32f411re nucleo board

Sreejasailu, 2024-10-26:
hi, i want to communicate between stm32f411re and raspberry pi4 via uart4. i want to flash the bin file onto the Nucleo board. but i am unable to find UART communication. and how can i communicate transmit and receive functions. and how can raspberry pi read and write commands. it would be more helpful if you provide me with the solution.
help me with the communication, thank you.
[image]

terjeio, 2024-10-27:
There is no UART4 in the STM32F411 chip so I am am not sure I understand what you want...
If enabling direct UART comms instead of serial over USB you have to desolder links SB13 and SB14 on the Nucleo, then you can connect the Pi UART to PA2/PA3. See section 6.8 in the [Nucleo64 documentation](https://www.st.com/resource/en/user_manual/dm00105823-stm32-nucleo64-boards-mb1136-stmicroelectronics.pdf).

Sreejasailu, 2024-10-28:
i am sorry it is not uart4, it is uart2. And i am again reframing my sentence as -
i want to use firmware which is the binary file which is downloaded from web builder. and from the firmware i get to flash onto the  board. and then from the firmware my machine works. but i want to receive data from raspberry pi 4, and that data has to be read by stm32 nucleo board. The g-code has to be generated and motors has to move to that particular location. i think i am clear. will this be possible, and how can i do this?
generally i have done communication both ways stm32 and raspberry pi4 through uart2, by writing send and receive functions in stm32cubeide. but here i cant change the firmware nor can i edit the binary file. for this what can i do ?

terjeio, 2024-10-29:
> for this what can i do ?

Easiest is to desolder SB13 and SB14, you can then communicate with the controller via PA2/PA3.
If you want or need to you can change the firmware by downloading and [compiling](https://github.com/grblHAL/core/wiki/Compiling-grblHAL) it in the STMCubeIDE.

Sreejasailu, 2024-10-29:
hey @terjeio, i have to download .zip file of the driver. in the drivers list when i opted under my board, i am getting f4xx.zip file which includes all the boards under f4xx. i am using stm32f411re board, so i want to only download the zip file of this board. 
And when i try to include this library in arduino ide. i am facing an error.
[image]
[image]

terjeio, 2024-10-30:
> i have to download .zip file of the driver.

This will not work since the driver relies upon a [number of submodules](https://github.com/grblHAL/core/wiki/Compiling-grblHAL#getting-the-source-code).

> so i want to only download the zip file of this board.

Then you have to download all the other required files separately as well...
The board to use for compilation is [selected here](https://github.com/grblHAL/STM32F4xx/blob/f288073d555fc38f1840703d3418f4208ac6bce4/Inc/my_machine.h#L22-L46), alternatively by setting a compiler symbol in platformio.ini [like this](https://github.com/grblHAL/STM32F4xx/blob/f288073d555fc38f1840703d3418f4208ac6bce4/platformio.ini#L144).

> And when i try to include this library in arduino ide.

The Arduino IDE is not supported for this driver, STM32CubeIDE is. Alternatively you may use [PlatformIO](https://github.com/grblHAL/STM32F4xx#quickstart) but I do not provide support for it.

Sreejasailu, 2024-11-01:
okay i understood. Thank you for the guidance. After i uploaded i am receiving errors continuously with the drivers, could you please take a look at it once.
../Inc/driver.h:49:10: fatal error: grbl/driver_opts.h: No such file or directory
   49 | #include "grbl/driver_opts.h"
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
what to be done to this, how to solve this and all? please guide me.

terjeio, 2024-11-07:
Your _grbl_ folder is empty? If so you have to [dowload the submodules](https://github.com/grblHAL/core/wiki/Compiling-grblHAL#getting-the-source-code).

Sreejasailu, 2024-11-09:
i have imported the project file, and been to all the files. i have seen that for the communication to other controller, there is spi, usb, i2c. in spi.h file, i have spi_write function, can i use this in main.c file?
i also want to know what all to keep in mind while editing the files, and all to be edited to communicate with raspberry pi 4.

terjeio, 2024-11-10:
> in spi.h file, i have spi_write function, can i use this in main.c file?

main.c should not be modified, add [a plugin](https://github.com/grblHAL/Templates) instead. If you want to use SPI for communication you will have to add your own protocol on top of SPI, better to use an UART port that supports the Grbl communication protocol. I2C has support for limited functionality via [this](https://github.com/grblHAL/Plugin_keypad/blob/master/keypad.c) or [this](https://github.com/grblHAL/Plugin_keypad/blob/master/display/i2c_interface.c) plugin, the latter can be used with this [protocol parser](https://github.com/grblHAL/grblHAL_MessageParser) on the Pi. For more advanced communication via I2C you will have to add your own protocol. Serial over USB is not available for the Nucleo64 boards unless it is via the ST-Link bridge, it works the same way as communication via UART (it actually is UART code running in the MCU).