grblHAL/core #618: A axis rotational issues

Fastbass, 2024-11-04:
2 issues I am having, machine is a CNC router with an open builds black box x32 controller. 1st issue is when soft limits are enabled it will only rotate negative, trying to rotate positive alarms that it's up against soft limits, $133 was defaulted to 360 degrees, open builds staff said to set at 0 degrees... It made no change to the operation. 2nd issue is when running a tool path that calls for part to be indexed 180 degrees or any value off of zero, axis rotates continuous despite the command in g code giving it a specific value. Peter at open builds suggested I post here since we are at a loss at this moment. Thanks in advance for any and all assistance.

terjeio, 2024-11-07:
Post your settings please.

Fastbass, 2024-11-07:
$0=10.0  ;  Step pulse time, microseconds
$1=255  ;  Step idle delay, milliseconds
$2=0  ;  Step pulse invert, mask
$3=0  ;  Step direction invert, mask
$4=0  ;  Invert step enable pin, boolean
$5=7  ;  Invert limit pins, boolean/mask
$6=1  ;  Invert probe pin, boolean
$9=1  ;  PWM Spindle as bitfield where setting bit 0 enables the rest
$10=511  ;  Status report options, mask
$11=0.010  ;  Junction deviation, millimeters
$12=0.002  ;  Arc tolerance, millimeters
$13=0  ;  Report in inches, boolean
$14=0  ;  Limit pins invert, mask
$15=0  ;  Coolant pins invert, mask
$16=0  ;  Spindle pins invert, mask
$17=0  ;  Control pins pullup disable, mask
$18=0  ;  Limit pins pullup disable, mask
$19=0  ;  Probe pin pullup disable, boolean
$20=0  ;  Soft limits enable, boolean
$21=0  ;  Hard limits enable, boolean
$22=1  ;  Homing cycle enable, boolean (Grbl) / mask (GrblHAL)
$23=0  ;  Homing direction invert, mask
$24=250.0  ;  Homing locate feed rate, mm/min
$25=1500.0  ;  Homing search seek rate, mm/min
$26=250  ;  Homing switch debounce delay, milliseconds
$27=5.000  ;  Homing switch pull-off distance, millimeters
$28=0.100  ;  G73 retract distance, in mm
$29=0.0  ;  Step pulse delay (ms)
$30=1000.000  ;  Maximum spindle speed, RPM
$31=0.000  ;  Minimum spindle speed, RPM
$32=0  ;  Laser-mode enable, boolean
$33=5000.0  ;  Spindle PWM frequency
$34=0.0  ;  Spindle off Value
$35=0.0  ;  Spindle min value
$36=100.0  ;  Spindle max value
$37=0  ;  Stepper deenergize mask
$39=1  ;  Enable printable realtime command characters, boolean
$40=0  ;  Apply soft limits for jog commands, boolean
$43=1  ;  Homing passes
$44=4  ;  Homing cycle 1
$45=3  ;  Homing cycle 2
$46=0  ;  Homing cycle 3
$47=0  ;  Homing cycle 4
$62=0  ;  Sleep Enable
$63=3  ;  Feed Hold Actions
$64=0  ;  Force Init Alarm
$65=0  ;  Require homing sequence to be executed at startup
$70=7  ;  Network Services
$73=1  ;  Wifi Mode
$74=  ;  Wifi network SSID
$75=  ;  Wifi network PSK
$100=199.55999  ;  X-axis steps per millimeter
$101=199.37998  ;  Y-axis steps per millimeter
$102=199.67999  ;  Z-axis steps per millimeter
$103=26.66480  ;  A-axis steps per degree
$110=2500.000  ;  X-axis maximum rate, mm/min
$111=1200.000  ;  Y-axis maximum rate, mm/min
$112=1500.000  ;  Z-axis maximum rate, mm/min
$113=2000.000  ;  A-axis maximum rate, deg/min
$120=75.000  ;  X-axis acceleration, mm/sec^2
$121=75.000  ;  Y-axis acceleration, mm/sec^2
$122=50.000  ;  Z-axis acceleration, mm/sec^2
$123=75.000  ;  A-axis acceleration, deg/sec^2
$130=735.828  ;  X-axis maximum travel, millimeters
$131=1244.000  ;  Y-axis maximum travel, millimeters
$132=135.000  ;  Z-axis maximum travel, millimeters
$133=36000.000  ;  A-axis maximum travel, degrees
$320=grblHAL  ;  Hostname, max: 64
$322= ;  IP Address
$323=  ;  Gateway
$324=  ;  Netmask
$325=23  ;  Telnet port
$326=80  ;  HTTP port
$327=81  ;  Websocket port
$337=  ;  WiFi Access Point (AP) BSSID
$341=0  ;  Tool Change Mode
$342=30.0  ;  Tool Change probing distance
$343=25.0  ;  Tool Change Locate Feed rate
$344=200.0  ;  Tool Change Search Seek rate
$345=200.0  ;  Tool Change Probe Pull Off rate
$346=1  ;  Restore position after M6 as boolean
$370=4  ;  Invert I/O Port Inputs (mask)
$376=0  ;  Rotational axes as bitfield
$384=0  ;  Disable G92 Persistence
$394=4.0  ;  Spindle on delay in s, range: 0.5 - 20
$396=30  ;  WebUI timeout in minutes
$397=0  ;  WebUI auto report interval in milliseconds
$398=100  ;  Planner buffer blocks
$481=0  ;  Autoreport interval in ms
$484=1  ;  Unlock required after E-Stop as boolean
$486=0  ;  Unlock required after E-Stop as boolean
$650=0  ;  File systems options as bitfield
$673=1.0  ;  Coolant on delay in s, range: 0.5 - 20

terjeio, 2024-11-07:
Your $133 setting is not 0: `$133=36000.000 ; A-axis maximum travel, degrees`

Fastbass, 2024-11-07:
I tried at "0" same results

Fastbass, 2024-11-08:
changed $376 to 1 and reset $133 to 0, with soft limits engaged A axis will only rotate negative, trying to go positive its says it will hit soft limits

Fastbass, 2024-11-08:
with soft limits enabled once A gets to 0 degrees on the DRO it wont rotate either direction due to soft limits

terjeio, 2024-11-08:
I am not able to replicate this behaviour, using the very latest version from a Web Builder build. Tested with both OpenBuilds Control and ioSender.

>  trying to go positive its says it will hit soft limits

What is the exact message?

Are you running a gcode file or sending gcode manually from the MDI?
If running a file can you add it to a comment?

Which version (build date) of grblHAL are you running? It is in this line in the console if using OpenBuilds Control:

`[05:53:25] [ $I ] [VER:1.1f.20241107:]`

Fastbass, 2024-11-08:
ver:1.1f..20240402
the problem exists when running the axis from the control screen jog buttons. when running gcode when a command line calls for A axis rotation, ex "A180.00" the axis starts rotating and continues non stop, must stop job to stop axis rotation.

Fastbass, 2024-11-08:
alarm msg reads "Unable to jog toward A+, will hit soft limits" or "unable to jog towards A-, will hit soft limits". With soft limits disabled I am able to jog either direction just fine, but when gcode is loaded and program rotates axis it will not stop at absolute or incremental positions, just rotates continuous.

terjeio, 2024-11-08:
> ver:1.1f..20240402

Please update to the latest version, I am not going to revert back to  yours for debugging an issue I am not able to replicate. You may use the [Web Builder](http://svn.io-engineering.com:8080/?driver=ESP32&board=BlackBox%20X32) to create the firmware.

Alos check that your sender is up to date since the messages does not come from the controller.