grblHAL/core #622: Z moving when not asked

Toutoune86, 2024-11-08:
I have a very weird issues repeating crashes with Z diving. I really see no way how this could happen, since it occurs in lines containing no supposed Z moves, both g1 or g2.
[poulie ralentie pour test3.txt](https://github.com/user-attachments/files/17681856/poulie.ralentie.pour.test3.txt)

It happens from line70 one it keeps going down at each line.
That is really weird and depressing, after so much time solving other issues, please could someone have clue or an idea about the cause of this ?

bLouChip, 2024-11-08:
I believe this effect you see is caused by your torch float distance in the post processor custom options being wrong.  See the comments I just made in sheetcam forum: https://forum.sheetcam.com/t/grbl-plasma-w-thc-torch-touchoff-and-rotary/7342/70

Toutoune86, 2024-11-09:
Ok, it was a stupid mistake from me. As I am currently focusing on small tubes cutting, the THC is not so necessary, so I  had tried without insisting to implement it  and failed (some people here gave me clue I still have to test, thanks) but I forgot my THC was still on in my grblHAL settings. I had to realize this issue with my moves did happen only after M3 and before any M5, which led me to realize the cause of these plunging. So I switched the THC setting off (as for now) and can say the issue is solved. 
The conclusion is, it is important to keep focused on what we are doing and think twice about the settings we pick !

closed 2024-11-09 by Toutoune86