grblHAL/core #630: when engraving an image in stucki mode the laser does not turn off during overscanning

distebia, 2024-12-01:
Hi, I use esp32 with mks dlc32, grblhal with latest firmware installed...if I work an image in grayscale mode no problem but if I work the same image in stucki mode, or any other image mode, I get small lines beyond the image, both on the right and left...it's as if the laser doesn't turn off in stucki mode, but my settings are only on at 90% and 0%...at zero the laser shouldn't draw these lines

terjeio, 2024-12-11:
What is "stucki mode" mode?

distebia, 2024-12-11:
stucki mode is a series of points close or less performed at the preset power, they are s900 and s0 if set to 90%...it is different from the grayscale where the S value varies to obtain the light and dark tones

[image]

distebia, 2025-01-30:
[image]

It often happens to me that with grblhal firmware the images come out like this, it's like the laser doesn't turn off in the section beyond the image, my $30 is 1000 and $31 is zero

distebia, 2025-02-14:
@terjeio what should I change to avoid having these extra lines ruining the work done?

terjeio, 2025-02-14:
I've just got around to installing LightBurn, now I have to learn to use it so that I can generate gcode to test with.
My CO2 laser is in the workshop some distance away, I have to go there to test and eventually debug. And I have to update the controller to a more recent version of grblHAL, due to spending far too much time on coding I have not had time to use my own machines lately...

If you can share a project (not gcode) which exhibit the issue then that would be helpful. And `$I` and `$$` output from the controller if you can as well.

distebia, 2025-02-14:
ok, i'll do it asap, is a lightburn lbrn2 file ok too?

terjeio, 2025-02-14:
> is a lightburn lbrn2 file ok too?

Yes - when I try to save my first project it defaults to .lbrn2.

distebia, 2025-04-08:
[prova.zip](https://github.com/user-attachments/files/19646883/prova.zip)

terjeio, 2025-04-09:
Is this file properly set up for generating the gcode?
If so the feed rate might be too high if you have a DC excited CO laser tube, they typically have a response time of ~1ms.
333mm/sec = 0.3mm/ms, so minimum pixel length should be 1 mm to be within tube response times. Actual is 0.08 mm if I read the gcode correctly. The overscan time is around 140ms (with my X-axis settings) when scoping the signal (triggered on direction changes), most of the time the spindle on/off signal is off during this time but not always like it should be. From what I can see this is likely due to incorrect gcode:
```
G1 X0.32S0:100:100:100
G1 X8.333
G1 X0.08Y-0.08
G1 X-8.333S0
```
the `G1 X8.333` motion is with the laser on (`S100`), it should be like this - which it is most of the time:
```
G1 X0.32S100
G1 X8.333S0
G1 X-0.08Y-0.08
G1 X-8.333S0
```
I do not know if this is a LaserBurn bug or there is a setting for this.

distebia, 2025-04-09:
I noticed that in lightburn there is an option to use g0 during overscan, it was enabled, I don't know if this affects it, I think not. The lb_cluster plugin was enabled..what can I do to eliminate the problem?
should the G1 X8.333 command have the s0 command or not?

distebia, 2025-04-10:
I took the same photo at the same speed and same settings, but with LB_CLUSTER disabled and the overscan problem does not appear