grblHAL/core #632: Building firmware for RP2040 fails

ahmadmadary, 2024-12-04:
Hi

I tried to build firmware for RP2040 board but it always fails (build failed:  503). what can be the cause of it?

[image]
### Discussed in https://github.com/grblHAL/core/discussions/203

<div type='discussions-op-text'>

<sup>Originally posted by **terjeio** October 17, 2022</sup>
I have started tinkering with a Web intreface, focus has been on generating the needed compiler symbols, not the UI layout.
A  [preview is available here](http://svn.io-engineering.com:8080/), currently symbols are output to the browser console only.

I have set up a development machine with the latest Ubuntu release which has platformio and docker installed. Next steps will be to post the compiler symbols to this machine to get the binary generated. It has been a while since I worked with browser apps/backends so any input on how to handle this will be appreciated.

Some technical details:

The browser code is pure javascript, no framework used (yet?).
A ["global" json file](http://svn.io-engineering.com:8080/drivers.json) will be hosted in the core repository, this contains links to all the drivers and info about plugins etc.
Each driver repository linked to will have [a json file](http://svn.io-engineering.com:8080/STM32F4xx/driver.json) containing information about the driver and the boards/controllers supported.
The file format of the json files are work in progress, but the current iteration seems promising considering the complexity of options available.

Finally: a ton of testing is required to to get this off the ground and performing smoothly. Help is needed as I do not own all the supported boards and thus I am not able to verify binaries for these.
</div>

terjeio, 2024-12-04:
The backend was down, restarted now.

closed 2024-12-05 by ahmadmadary

jattie-ire, 2025-01-30:
I have the same error again, unable to build and Pico configs.

terjeio, 2025-01-30:
backend down again... up now.