grblHAL/core #634: Cant use ESP32 driver when wifi is enabled

DLR3D, 2024-12-04:
Hello, I am using the latest version of the grblHAL for ESP32, using a lolin esp32 board.

There is a issue that's impeding me from continuing the configuration, whenever wifi and webui is enabled the grbl does not load. The web interface is active and I can see setings and files but no commands receive response and no stepping signal is generated, also the serial interface keeps printing these errors over and over:

<ESC>[0;31mE (241483) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:<ESC>[0m<CR>
<ESC>[0;31mE (241483) task_wdt:  - IDLE (CPU 1)<ESC>[0m<CR>
<ESC>[0;31mE (241483) task_wdt: Tasks currently running:<ESC>[0m<CR>
<ESC>[0;31mE (241483) task_wdt: CPU 0: IDLE<ESC>[0m<CR>
<ESC>[0;31mE (241483) task_wdt: CPU 1: tiT<ESC>[0m<CR>
<ESC>[0;31mE (241483) task_wdt: Print CPU 0 (current core) backtrace<ESC>[0m<CR>
<CR>
<CR>
Backtrace:0x40176D3B:0x3FFB1140 0x400853AD:0x3FFB1160 0x4017A263:0x3FFBF8A0 0x40176ABB:0x3FFBF8C0 0x4008D155:0x3FFBF8E0 0x4008EFB1:0x3FFBF900 <CR>
<CR>
<ESC>[0;31mE (241483) task_wdt: Print CPU 1 backtrace<ESC>[0m<CR>
<CR>
<CR>
Backtrace:0x40089C05:0x3FFB1740 0x400853AD:0x3FFB1760 0x4000BFED:0x3FFCBB50 0x4008F21E:0x3FFCBB60 0x4008C9AA:0x3FFCBB80 0x4011C36B:0x3FFCBBC0 0x4010A1BC:0x3FFCBBF0 0x4010A273:0x3FFCBC10 0x4008EFB1:0x3FFCBC40 <CR>
<CR>

I have tried:
- Changing webui versions, same result
- Disabling SD card, same result
- Disabling Telnet, same result
- Disabling FTP, same result
- Disabling wifi completely, the board boots correctly and responds and generates steps via serial. No watchdog errors show up

Can anyone help me solve this problem as I am at a loss of what to try next to fix this.

terjeio, 2024-12-04:
Which board did you build for?

DLR3D, 2024-12-04:
I used the online builder and built for ESPDUINO.

I also tried building it manually for this board, the result is the same.

terjeio, 2024-12-04:
> the only problem are the pin definitions that are completely wrong.

Completely wrong in relation to what?

> If I try compiling using PlatfomIO the error appears once again

Which SDK version are you using? The Web Builder uses 4.3.2.

DLR3D, 2024-12-04:
> Completely wrong in relation to what?
Wrong in relation to what I needed from this board. Sorry, I wasn't clear with the use case. I am using a custom board so I need to redefine the default pinout.

> Which SDK version are you using? The Web Builder uses 4.3.2.
I used the default version that was defined on the platformio.ini config 4.0, I will try with that version and report back.

DLR3D, 2024-12-04:
Managed to fix the issue by using Linux virtual machine and compiling using the recommended method using the files I had previously changed. The board is now working as it is supposed to.

There is still and issue to be solved, there is some error in the platform.io configuration in your files that makes compiling using it impossible, it would be amazing to fix this as it would allow multi platform usage with almost no configurations for users, any way I can help debug the issue?

msepulve, 2025-01-05:
Hi! 
I’m experiencing exactly the same problem when trying to use compiled version on my MKS TinyBee. Web builder versions work just fine.
Compilation as per the GitHub instructions work fine, upload, vie settings via usb terminal Ok. Problem start immediately after a device connects to the AP:
Domingo, can you share what did you do to compile a working version?

DLR3D, 2025-01-06:
Hello, I managed to compile a working version by compiling on linux using
the recommended packages, my problem was caused due to some incorrect
espressif library being downloaded on platformIO which caused the incorrect
compilation. I haven't spent more time looking into the issue since then

msepulve, 2025-01-06:
Obrigado Domingo. Please tell us the library that was causing you the problem? I have been struggling with the same issue for days but not seeing/recording any error during the compilation I wouldn’t know where to start. I also compile form platformio command line and have been looking for solutions changing settings in the sdkconfig files to no avail. Look forward to know your solution, thanks in advance

msepulve, 2025-01-06:
Thanks to Domingo inspiration, I moved to a brand new updated Ubuntu 24 installation and installed everything afresh, following grblHAL GitHub instruction step by step. Then created a platformio.ini for my MKS TinyBee, using the settings as per the Web Builder. But the results are the same as what Domingo encountered; as soon as a device connects to the AP, the cycling messages star showing up. Please someone have any ideas about this?