grblHAL/core #636: MPG mode don't start

MrKapim, 2024-12-06:
Hi Terjeio, I downloaded the new grblHAL version 20241128 (I'm was using 20240205) to update my machine and now I have errors to communicate...  My machine have a H100 spindle on serial port 2 and is in use a MPG on serial port 1. The problem is the MPG don't claimed the serial anymore. The $pins show the claimed ports and don't show the MPG, only the ModBus and the others ports free (USART 2 and USART 3). I missing anyone or I make mistakes on my configuration????

MrKapim, 2024-12-06:
In time...  My drive is a stm32H743... for MB SKR3 EZ

dresco, 2024-12-06:
If you can share your configuration, will see whether I can replicate on my SKR3 here?

In the meantime, I have an H7 board configuration with 2x UART, and can confirm they are both being claimed as expected. My board config  is [here](https://github.com/dresco/STM32H7xx/blob/2d3a354bd89e8e2b3a6094a1a5502c3103433ea1/boards/dresco_octave_map.h#L33-L42).

MrKapim, 2024-12-07:
Hi @dresco, my board (stm32H743) using the firmware 20240205 communicate normally with MPG and spindle H100  (serial ports 1 and 2) and the $pins show the serial 3 free but after update to firmware 20241127 only one serial is claimed. I didn't change anything in the configuration.  Here are some images...
[image]
[image]
[image]
[image]
[image]
[image]
[image]

MrKapim, 2024-12-07:
[image]

terjeio, 2024-12-07:
> I missing anyone or I make mistakes on my configuration????

You recently added the H100 spindle? If so Modbus it will claim the first free Modbus capable port (stream) if `MODBUS_RTU_STREAM` is not defined in the map file. This will be the first free port for most drivers, 0 or 1 depending on USB beeing used as the main communication channel, but not all.
By default the MPG is set to claim port 0 via the `MPG_STREAM` definition, when this is claimed by Modbus then the claim will fail. If it fails then this means you will have to define which port to use for either one or both streams in your map file or in _my_machine.h_.

Depending on your wiring add `#define MODBUS_RTU_STREAM 1` or `#define MPG_STREAM 1` to fix this.
Note if you want to add `#define MPG_STREAM 1` in your map file then you have to add `#undef MPG_STREAM` on the line before it.

MrKapim, 2024-12-08:
Ok Terje, I did what you pointed out above and the results is NOK. I make the changes (#define MODBUS_RTU_STREAM 1 and #define MPG_STREAM 1) in my_machine.h. OK, the serial 2 claimed by modbus but the serial 1 not claimed. I uncommented #define MPG_ENABLE 1 together, NOK. I moved #define MPG_STREAM 1 to map file and #undef MPG_STREAM before, NOK. The serial 1 is not claimed. But I uncommented the #define KEYPAD_ENABLE 2 (together with MPG_ENABLE and alone) and the serial 1 was claimed.  Pics to evaluate...
 
[image]
[image]
[image]
[image]
[image]

MrKapim, 2024-12-08:
The MPG is not claimed. Any other ideas?

terjeio, 2024-12-08:
> I make the changes (#define MODBUS_RTU_STREAM 1 and #define MPG_STREAM 1) in my_machine.h.

You cannot assign the same stream for both. Set one to 0 and the other to 1, or only set one to 1.

MrKapim, 2024-12-09:
Hi Terjeio. You is the guy!!!   After fixed my errors (same number for streams), works normally. Thanks a lot!!!
And thanks Jon, you all time is the first to communicate me about suggestions to fix the errors.
I have a question...  My A-axis (rotational), after 360 degrees continues showing your position. I think it should show 0 again and whenever it reaches 360, start from 0 again. Same negative side. Can you change this???

MrKapim, 2024-12-09:
And other question...  In my case, my set has a 6:1 reduction and I've seen other reductions on the market, 4:1, 10:1. Would it be possible to include this in the firmware about rotational axes? Because it is showing machine position 6x the real position (part)

MrKapim, 2024-12-09:
Before you talk about the steps, I put the correct steps to show the final position but hurt the speed...

terjeio, 2024-12-09:
> I think it should show 0 again and whenever it reaches 360, start from 0 again. Same negative side. Can you change this???

No, I cannot. You may customize the real-time report if you want to but there may be unwanted side-effects if you do.

BTW there is a new setting for enabling a trick for [fast rewinds](https://github.com/grblHAL/core/blob/master/changelog.md#20240719).

> ... I put the correct steps to show the final position but hurt the speed...

If feedrate is too slow when both linear axes and the rotary is commanded to move at the same time then this is a known issue. A workaround for this is to enable the [rotary feed fix](https://github.com/grblHAL/core/discussions/241).