grblHAL/core #638: M6 and returning to old position

tcurdt, 2024-12-16:
I have a grblHAL

```
$I
[VER:1.1f.20240907:]
[OPT:NMZHS,100,1024,3,0]
[NEWOPT:ENUMS,RT+,HOME,ES,TC,SED,RTC]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:218K]
[DRIVER:RP2040]
[DRIVER VERSION:240907]
[DRIVER OPTIONS:SDK_1.5.1]
[AUX IO:3,2,0,0]
```

and I have configured grblHAL as such:

[image]

[image]

Now when I run this gcode:

```
G21
G90
G0 X50 Y50 Z10 F6000
G1 X100 Y50 Z10 F2000
M6
G1 X0 Y0 Z10 F2000
G1 X50 Y0 Z10 F2000

```

When I jog around during the tool change it will NOT restore the original position - not even in Z.
And so the cut `G1 X0 Y0 Z10 F2000` comes out wrong.

While this feels like a bug, I suspect I might just be me holding it wrong.

Or maybe I have some wrong expectations?

tcurdt, 2024-12-17:
Unfortunately ioSender/Windows is not an option.

AFAIK gSender now has grblHAL support - but maybe something is missing. 
The modes are at least configurable via firmware settings through gSender

As the M6 is passed through by the sender, and the sender just resumes, it let me to believe the problem lies with grblHAL.
But if it's more complicated than that, that assumption might be of course flawed.

What I am wondering is why is a "restore position after M6" sender dependant?

So [this is the toolchange protocol](https://github.com/grblHAL/core/wiki/Manual-tool-change-protocol), right?

https://github.com/user-attachments/assets/a96708fe-a783-4c6d-b320-6a540a43ef36

To me it looks like the sender is detecting tool mode correctly.
And aren't I giving the ack via pressing resume?
Could I try to send the required gcode via console for testing maybe?

terjeio, 2024-12-17:
> What I am wondering is why is a "restore position after M6" sender dependant?

If the sender is not interacting correctly with the tool change protocol then the tool change settings are meaningless.

> So [this is the toolchange protocol](https://github.com/grblHAL/core/wiki/Manual-tool-change-protocol), right?

Yes.

> And aren't I giving the ack via pressing resume?

I do not know, ioSender does it automaticially when the reported state transitions to `Tool`. And then suspends sending data from the loaded gcode file. In the video above I do not see any transitions to the `Tool` state, only to `Running` and `Jogging` (from `Idle`).

> And aren't I giving the ack via pressing resume?

Again, I do not know. The first ACK is character code 0xA3, the second is 0x7E or `~` (cycle start). The sender should also resume if the cycle start signal is received from the controller.

> Could I try to send the required gcode via console for testing maybe?

If you are able to send the first ACK character code (0xA3) in response to the `Tool` state transition then yes. I do not know how to do that from the keyboard though and if the sender allows it - perhaps `ALT 1 6 3` or `ALT 0 1 6 3`? It might be easier to do this from a terminal program, especially so if the sender interferes in the communication. Or if you are compiling locally change the the tool change ACK character to a printable one [here](https://github.com/grblHAL/core/blob/50a7507f6fe98fa22c062685f46e9d06d20e85f8/grbl.h#L131), e.g. `'&'`?

gSender has its own internal tool change support, perhaps that is your only option for now?

---

If you have a Pico W with a SD card connected then you may try the [WebUI](https://github.com/grblHAL/Plugin_WebUI/), it supports the tool change protocol.

tcurdt, 2024-12-17:
> > What I am wondering is why is a "restore position after M6" sender dependant?
> 
> If the sender is not interacting correctly with the tool change protocol then the tool change settings are meaningless.

I wasn't sure how the whole tool change thing is happening. I thought that's all in grblHAL land and the M6 is blocking on the sender side. But it seems like it's more like a 2nd session or side channel.
Which makes it more clear this needs sender support.

But I am surprised it even works the way does if there is no full support for the tool change.

> gSender has its own internal tool change support, perhaps that is your only option for now?

You might speak some sad truth.
But given the other grief gSender is giving me I might have to look for yet another option.

> If you have a Pico W with a SD card connected then you may try the [WebUI](https://github.com/grblHAL/Plugin_WebUI/), it supports the tool change protocol.

I got the flexi board and no plans to switch.

Closing as this doesn't seem directly grblHAL related afterall.

Thanks for the help!

closed 2024-12-17 by tcurdt