grblHAL/core #644: Unexpected Laser Behaviour

engigeer, 2024-12-29:
While troubleshooting a new laser spindle driver I came across some unexpected behaviour and I was hoping to confirm if it this is intentional or perhaps a bug. My expectation was that in laser mode, the laser/spindle will only go to the ```on``` state (with rpm/power >0) when motion is occurring. However, it seems that it is possible to turn it on while stationary by simply issuing a```M03 S##``` command from the console. I verified this behaviour is not specific to my driver and also occurs with the RP2040 default PWM driver.

Perhaps this is intentional, and I do imagine there are some applications where the capability to fire the laser while stationary could be useful (e.g., a short dwell during laser cutting to pierce the workpiece). However, in this case I would note that with the current behaviour the laser does not resume if a feedhold is issued and cleared while the motion is stationary and the laser is on. It will only resume after a motion command is issued, which seems somewhat inconsistent.

terjeio, 2024-12-29:
> My expectation was that in laser mode, the laser/spindle will only go to the on state (with rpm/power >0) when motion is occurring. However, it seems that it is possible to turn it on while stationary by simply issuing aM03 S## command from the console. 

There is a bug in that the laser is switched on if the current motion mode is G0, it should only be switched on if not G0. I'll fix that. To avoid switching on the laser when no motion consider using `M4` instead of `M3`.

> However, in this case I would note that with the current behaviour the laser does not resume if a feedhold is issued and cleared while the motion is stationary and the laser is on. It will only resume after a motion command is issued, which seems somewhat inconsistent.

This is how legacy Grbl behaves as well, perhaps an inherited bug?

engigeer, 2024-12-29:
Thanks for the clarification.

On a slightly related note, the new spindle driver I am working on is intended to allow control of a laser connected to a daughter board via modbus. During testing I have noticed that the **(OFF / CW / CCW)** radio buttons in ioSender do not change state when I issue ```M03``` and ```M05```` commands from the console (although the commands seem to be parsed correctly by the firmware and the spindle state changes accordingly). Any suggestions on why this is might be occurring are greatly appreciated.

terjeio, 2024-12-30:
$10 is set to what? 511 is the default value.
Is the spindle state reported back in the `|A:` realtime report element? `S` for spindle on, `C` for counterclockwise.

engigeer, 2025-01-04:
> This is how legacy Grbl behaves as well, perhaps an inherited bug?

Yes, likely this is the case. There is one final quirk I have noticed about sending ```M03 SXXX``` in laser mode while stationary. It seems that sending any subsequent commands that includes a new ```SXXX``` value, will set the **RPM** value to zero. I think this is actually preferable for my use case, just somewhat unexpected, so I figured worth mentioning as I close out this issue in case it might help someone who is reading in the future.

> $10 is set to what? 511 is the default value. Is the spindle state reported back in the `|A:` realtime report element? `S` for spindle on, `C` for counterclockwise.

Thank you very much for the troubleshooting advice. It helped me realize that I had an error in my ```spindleGetState()``` function, so although the spindle speed was being set, the sender was not able to verify this when polling for the status. All seems to be working correctly now.

closed 2025-01-04 by engigeer

dresco, 2025-01-09:
Hi Terje, sorry have a question on this as well. (Testing with the latest core code)..

Was checking the PWM behaviour in preparation for putting a diode laser on my machine, and see some unexpected (to me) behaviour.

With the following gcode, the laser is enabled and disabled as expected with the first G1 move (using M4), however it remains enabled following the second G1 move (using M3). PWM output was manually stopped afterwards with the G0 command.

Is that the correct behaviour, must admit I was expecting the PWM to turn off at the end of the move?

```
M4 S750
ok
G1 X10 F1000
ok
G0 X0
ok
M5
ok
M3 S750
ok
G1 X10 F1000
ok
G0
ok
```

[image]

dresco, 2025-01-09:
> @dresco It is the same behaviour as legacy Grbl, at least on my Mega. And AFAIKT it is consistent with the [documentation](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode#laser-mode-operation).

Oh yes I see, that's  fair enough then. Thanks for looking!