grblHAL/core #655: Unable to compile simulator for Windows

abcurrie, 2025-01-08:
Error:

Command failed: make
/usr/bin/x86_64-w64-mingw32-ld: CMakeFiles/grblHAL_sim.dir/objects.a(machine_limits.c.obj):machine_limits:(.text+0x179c): undefined reference to `ffs'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/grblHAL_sim.dir/build.make:883: grblHAL_sim.exe] Error 1
make[1]: *** [CMakeFiles/Makefile2:85: CMakeFiles/grblHAL_sim.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

abcurrie, 2025-01-08:
I tried again with probe input disabled and got this:

Command failed: make
/home/terjeio/grblHAL/Simulator/src/grbl/nvs_buffer.c: In function ‘nvs_alloc’:
/home/terjeio/grblHAL/Simulator/src/grbl/nvs_buffer.c:316:35: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
316 | mem_address = (uint8_t *)((uint32_t)(mem_address - 1) | 0x03) + 1; // Align to word boundary
| ^
/home/terjeio/grblHAL/Simulator/src/grbl/nvs_buffer.c:316:23: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
316 | mem_address = (uint8_t *)((uint32_t)(mem_address - 1) | 0x03) + 1; // Align to word boundary
| ^
/home/terjeio/grblHAL/Simulator/src/grbl/machine_limits.c: In function ‘get_homing_rate’:
/home/terjeio/grblHAL/Simulator/src/grbl/machine_limits.c:660:15: warning: implicit declaration of function ‘ffs’ [-Wimplicit-function-declaration]
660 | idx = ffs(cycle.mask) - 1;
| ^~~

/usr/bin/x86_64-w64-mingw32-ld: CMakeFiles/grblHAL_sim.dir/objects.a(machine_limits.c.obj):machine_limits:(.text+0x179c): undefined reference to `ffs'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/grblHAL_sim.dir/build.make:883: grblHAL_sim.exe] Error 1
make[1]: *** [CMakeFiles/Makefile2:85: CMakeFiles/grblHAL_sim.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

terjeio, 2025-01-08:
I have a pending fix for this, add these lines to the end of [nuts_bolts.h](https://github.com/grblHAL/core/blob/3f37d8c47830ed2cde334bf376f3afea63d371ab/nuts_bolts.h#L236):

```
#ifdef _WIN32

static int ffs (int i)
{
    int idx = 0;

    while(i) {
        idx++;
        if(i & 1)
            break;
        i >>= 1;
    }

    return idx;
}

#endif // _WIN32
```

terjeio, 2025-01-11:
Fix committed.

closed 2025-01-11 by terjeio