grblHAL/core #661: Timing of Stepper ISR is different with ARPE-BIT at STM32F756

AndreLukas121, 2025-01-13:
Hi,

We put the grblHAL under a lot of stress by driving small travel distances at regular intervals. We noticed that the stepper ISR kept "waiting". 

This can be explained by the fact that the stepper ISR (ISR_CODE void ISR_FUNC(stepper_driver_interrupt_handler)(void) in stepper.c) writes the value for the auto-reload register itself and the ARR value is adopted immediately (see static void stepperCyclesPerTick (uint32_t cycles_per_tick) in driver.c). However, in some cases the execution time of the stepper ISR (in Picture 1 the blue arrow) takes longer than the calculated ARR Value 2 so that the counter from the timer counts up to the maximum value.

Picture 1:  "WAIT" for next event
[image]

A possible error correction with Auto-Reload Preload Enable (ARPE-BIT): 

The bit only changes the ARR value with the update flag at each overflow event. However, this also changes the timing.
In the next two images we can see the differences in timing. In the second ISR we change the ARR value 1 to ARR value 2. If ARPE = 1 this means that only the next ISR will be affected by the change and if ARPE = 0 (current implementation), the ISR that is still running. differ between the timing in the second ISR we change the ARR Value to ARR-Value 2. This influenced the next ISRs.

Picture 2: ARPE-Bit = 0
[image]

Picture 3: ARPE-Bit=1
[image]

Finaly, we a litte bit confused what the actual behavior should be  and whether it has an impact on repetition accuracy. 
And if the ARPE bit isn't a solution, does anyone have an idea how to solve it instead?

 Can someone help us. Thank you in advance.

Andre

terjeio, 2025-01-23:
> However, in some cases the execution time of the stepper ISR (in Picture 1 the blue arrow) takes longer than the calculated ARR Value 2 so that the counter from the timer counts up to the maximum value.

Has this been observed? If so with which settings and gcode? And what is the symptom when there is unexpected behaviour?
The timer is used in count down mode so should never count outside 0 to the latest ARR value set.
I have occasionally seen very long step intervals coming from the core, this is why there is an [upper limit](https://github.com/grblHAL/STM32F7xx/blob/0cb4cef56b361ea772347b1a81d8aaa37832810c/Src/driver.c#L527) imposed in the driver. But I doubt this is an issue - at least noone has reported anything related to it.

> And if the ARPE bit isn't a solution, does anyone have an idea how to solve it instead?

Lower the step rate? Step pulse width too long and should be reduced?
If the step rate is too high the driver crashes due to reentering the interrupt handler, this can be detected and should perhaps result in halting the job. I can run 3 axes with step rates > 300KHz on the bench, this with a release build (-O2). If testing with a debug build the max rate is significantly lower, around 100KHz it seems.

terjeio, 2025-02-11:
> The motor runs and sometimes the motor stops for about 39 seconds and the stop time of 39sec is constant

To count from 0 to overflow in 39s implies a timer clock of ~110Mhz, however it runs at 27MHz.

> The GCode is generated automatically with this function.

So this means it happens randomly, not every time the same move is executed?

> Can you please show where the config bit for the countdown is set.

[Here](https://github.com/grblHAL/STM32F7xx/blob/6cc779cde5fb521c3a4c8ae0e987173cac9b6db3/Src/driver.c#L2336).

AndreLukas121, 2025-02-11:
_To count from 0 to overflow in 39s implies a timer clock of ~110Mhz, however it runs at 27MHz._

A: Yes, CLK-freuqency is 108MHz. 

_So this means it happens randomly, not every time the same move is executed?_
A: Yes, it happens randomly. 

Thanks, we are using the GRBL_Build: 20220906 There is no definition of the countdown mode there. So I think that in newer builds there are no problems in that direction.

closed 2025-02-11 by AndreLukas121