grblHAL/core #662: Motors not powered, driver issue?

raoullector, 2025-01-16:
Hi,

I’m not sure if this is the right section, but I feel like my issue might not be related to the drivers themselves. I have a 4-axis CNC machine with one coupled axis, controlled by an SKR 3 (STM32H723) using EZ2209 drivers. I’m running GrblHAL, using either versions generated with the WebBuilder or compiled myself (STM32CubeIDE). However, I decided to replace my five EZ2209 drivers with EZ5160 Pro drivers. From that point on, nothing works anymore :

1) I install the new EZ5160 drivers, recompile the firmware to work with them, and power on the machine. Result: none of the steppers are powered, and M122 returns "M122 => MSG: Warning: Could not communicate with stepper driver!"

2) I try all the $RST versions. Result: it still doesn’t work.

3) I remove the drivers one by one in case one of them is defective. Result: still out of work.

4) I decide to revert to the previous working configuration, reinstall the EZ2209 drivers, and reload the old firmware. Result: fxxk! It works no more! The only small positive point is that the motors are slighty powered, but I can't change the current nor can I  communicate with the drivers. Just like with the EZ5160, M122 returns "M122 => MSG: Warning: Could not communicate with stepper driver!"

I feel like the different firmware updates have conflicted with each other and are causing interference. For example, I noticed that the value of the BUILD_INFO symbol, which is supposed to display with $I, did not always match what I had specified during compilation.

What do you think about this, considering that what I want to achieve is getting the EZ5160 to work?

dresco, 2025-01-17:
I've got the EZ5160 drivers installed on my SKR3 (H743 variant). Can't send you my exact firmware because of the processor difference, but if you let me know what options you want, can build/send you a H723 binary for testing?

Couple of things I'd suggest for troubleshooting;

Enable just one Trinamic axis at a time with the `$338` axis mask. grblHAL will attempt to communicate with all drivers enabled there, and I believe will error if any fail to respond.

People have seen issues before where the drivers weren't initialised in time for the initial configuration attempt (I guess related to power supply startup - had never been able to recreate locally). I *think* Terje increased the delay before initialisation to account for this, but perhaps worth powering the board first, and then resetting the controller to see if there is any difference in behaviour?

dresco, 2025-01-17:
> Indeed, powering the board and then performing a reset of the board ($RST=*) and the motors (M122I) solved the problem.

Oh, I was thinking of the reset button on the board, just to restart the software initialisation after the drivers had time to settle, but whatever works ;)

I guess we still need a longer delay before the initial configuration attempt, @terjeio any thoughts on this?

raoullector, 2025-01-17:
After several experiments, the behavior seems random:
- sometimes everything works on the first try, immediatly after the board is powered,
- other times a physical reset (on the board) after powering up is enough,
- and unfortunately, quite often it takes multiple combinations of physical resets and logical driver resets ($RST=&).

In practice, this is very inconvenient because it would require adapting the sender to these changing behaviors...

dresco, 2025-01-17:
> logical driver resets ($RST=&)

Must admit, am not sure what that would be doing to help, other than a soft reset after settings are restored. You could maybe also just try the soft reset (Ctrl-C from a terminal)?

Is hopefully all the same root cause of trying to configure the drivers before they're ready.

terjeio, 2025-01-17:
> For example, I noticed that the value of the BUILD_INFO symbol, which is supposed to display with $I, did not always match what I had specified during compilation.

`BUILD_INFO` is a default value for the build info _setting_, it will not (and can not) be updated on an update. To update the setting after reflashing all settings has to be reset to default by `$RST=*`. BTW at any time the string can be changed by `$I=<my build string>` unless the config for blocking build string updates is set.

>> logical driver resets ($RST=&)

> Must admit, am not sure what that would be doing to help, other than a soft reset after settings are restored.

This is not needed unless one want to undo any changes since the last settings reset.

> You could maybe also just try the soft reset (Ctrl-C from a terminal)?

A soft reset will not reinitialize the drivers, only `M122I` will.

The power up delay is currently 100ms plus the time it takes to boot up most other subsystems. It can be made longer if this is the issue. How is the system powered up? A single switch powering the motors and the controller at the same time? Or switched on manually with separate switches?

There is a [template plugin](https://github.com/grblHAL/Templates/blob/master/my_plugin/Motor_power_monitor/my_plugin.c) that _could_ be extended  delay controller bootup (or leave it in alarm 17) until the motors are powered. A free aux input is required for that plus compatible power up signal delivered from the motor supply.

terjeio, 2025-01-17:
Delay is [here](https://github.com/grblHAL/Plugins_motor/blob/431145a34e230c20c20369051850ed3a17bc0b8c/trinamic.c#L2424).

raoullector, 2025-01-17:
> BUILD_INFO is a default value for the build info setting

Alright, got it.

> How is the system powered up? A single switch powering the motors and the controller

Exactly. The SKR3 controller integrates and powers the EZ5160 drivers itself.

> There is a [template plugin] that could be extended [...]. A free aux input is required for that plus compatible power up signal

Seems an idea, but where can I find the power-up signal since the drivers are plugged on the board ?
And, on the other hand, since I assume that the board itself is capable of determining whether the drivers are powered or not, why is an auxiliary input needed?

> Delay is [here]

To test, I multiplied the duration by 10 (hal.delay_ms(1000, NULL);), but it had no effect on the issue: the moment when the drivers are initialized (= motors powered on = M122 works) remains random.

For now, my workaround is to disable homing ($22=0) and then chain $RST=&, M122I, and hard resets until the drivers finally initialize. Not great...

terjeio, 2025-01-17:
> why is an auxiliary input needed?

As seen from the MCU - how else can it sense the outside world.

>  $RST=&,

Why do you have to reset the settings? Are they randomly changing values on boot up?

Could it be that the MCU is powered via USB - or a debug probe? Before motor power is on?

raoullector, 2025-01-18:
Hello,

> how else can it sense the outside world ?

In fact, in this specific case, I feel that Grbl could determine itself the initialization state of the drivers without relying on external information. For example, I imagine a thread could loop every 100 ms to check if drivers are enabled. It may look like:
```
if (driver_enabled.mask != trinamic.driver_enable.mask) {
trinamic_drivers_init(trinamic.driver_enable);
} else {
exit_thread();
}
```
What do you think of this idea?

> Why do you have to reset the settings? Are they randomly changing values on boot up?

No, you're right. Finally, the sequence that works best on my machine is this:
```
$22=0
$X
M122I
```
I repeat it until the drivers are fully initialized.

terjeio, 2025-01-18:
> I repeat it until the drivers are fully initialized.

IMO not going to work, at some point the controller has to be able to communicate that there is a problem - not be unresponsive forever.
So I am going to add a new overridable symbol `TRINAMIC_STARTUP_TIMEOUT` that I default to 5000 (ms), add it to the board map if a board needs a longer delay, to _my_machine.h_ or _platformino.ini_ or wherever else it can be defined.

And I may add a HAL entry point for querying or resetting stepper drivers, could be used to make plugin code more flexible.