grblHAL/core #666: Cant build new firmware for the BTT Rodent ESP32

enghamar, 2025-01-26:
I have a firmware from 14. dec which works, but is missing probe and stepper on/off, so I need to generate a new firmware for the board. I have attached the BIN files.

[20241214 BTT rodent.zip](https://github.com/user-attachments/files/18549585/20241214.BTT.rodent.zip)

I have tried generating new firmware files using the web builder, but no matter what combination of settings I cant get the board to work with the newer firmware.

Unfortunately I dont have a json file showing the web builder configuration on the working firmware.

I can see that there have been some changes made to the BTT Rodent board map 14. dec and 17. dec, but I dont know if it is related.

I have tried generating new firmware files since about 16-17. dec but have been unsuccesful, and after a lot of trial and error I am about to give up the the board and grblHAL and just go back to the arduino GRBL board I used before.

terjeio, 2025-01-26:
> I have tried generating new firmware files since about 16-17. dec but have been unsuccesful,

Compilation failed? If so I cannot remember it beeing reported before, and I just built a version that starts up:
```
GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:Warning: Could not communicate with stepper driver!]
[MSG:Warning: Failed to initialize SD card]
[VER:1.1f.20250126:]
[OPT:VNMSL,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,ES,REBOOT,SED,RTC,WIFI,SD]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FLASH 4K]
[FREE MEMORY:168K]
[DRIVER:ESP32]
[DRIVER VERSION:241222]
[DRIVER OPTIONS:4.3.2]
[BOARD:BTT Rodent]
[WIFI MAC:e4:65:b8:19:d9:8c]
[IP:]
[PLUGIN:Trinamic v0.24]
[PLUGIN:WebUI v0.24]
[PLUGIN:SDCARD v1.19]
[PLUGIN:FS macro plugin v0.15]
ok
```

I do not have a Rodent board so I cannot verify if it is working or not. And I cannot see any attempt other than my own to build for the Rodent today.

enghamar, 2025-01-26:
Hi. Thank you for getting back to me so quickly about this.

The compilation of the firmware seems to work as intended, and I get BIN files, which I flash to the board using the esp32 flash download tool.

But I have tried to generate different firmwares over the last month, and haven't gotten any of them to work. When I try to use the board with newer firmware, I dont get any of the usual GRBL boot output I usually see in the console.

When I go back to the firmware from 14. Dec it works as intended.

How can I do to troubleshoot this issue?

terjeio, 2025-01-26:
>  I dont get any of the usual GRBL boot output I usually see in the console.

In a sender console? If so which?
If you are using Windows can you try connect with a terminal program such as putty? Any output in that?

enghamar, 2025-01-27:
Hi again.

I have attached different firmware files and the output I am getting from Coolterm from the board when booting them:

[20241214 known working.zip](https://github.com/user-attachments/files/18563599/20241214.known.working.zip)

Working "20241214 known working" using 24v PSU:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1820
ho 0 tail 12 room 4
load:0x40078000,len:12844
load:0x40080400,len:3020
entry 0x400805e4

GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:Warning: Failed to initialize SD card]
[MSG:Warning: Failed to initialize SD card]
[MSG:WIFI STA DISCONNECTED]
```

[20250127 BTT rodent with webUI.json](https://github.com/user-attachments/files/18563605/20250127.BTT.rodent.with.webUI.json)
[20250127 BTT Rodent with webUI.zip](https://github.com/user-attachments/files/18563604/20250127.BTT.Rodent.with.webUI.zip)

Not working "20250127 BTT rodent with webui" using 5v USB power:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1820
ho 0 tail 12 room 4
load:0x40078000,len:12844
load:0x40080400,len:3020
entry 0x400805e4
[MSG:Warnin
GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:Warning: Could not communicate with stepper driver!]
[MSG:Warning: Failed to initialize SD card]
```

Not working "20250127 BTT rodent without webui" using 24v PSU:

`........................................................................................`

[20250127 BTT Rodent without webUI.zip](https://github.com/user-attachments/files/18563602/20250127.BTT.Rodent.without.webUI.zip)
[20250127 BTT rodent without webUI.json](https://github.com/user-attachments/files/18563600/20250127.BTT.rodent.without.webUI.json)

Not working "20250127 BTT rodent without webui" using 5v USB power:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1820
ho 0 tail 12 room 4
load:0x40078000,len:12844
load:0x40080400,len:3020
entry 0x400805e4

GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:Warning: Could not communicate with stepper driver!]
```

When powering the board with a 24v PSU, I get no output from the terminal at all.

[20250126 BTT rodent.json](https://github.com/user-attachments/files/18563601/20250126.BTT.rodent.json)
[20250126 BTT rodent.zip](https://github.com/user-attachments/files/18563603/20250126.BTT.rodent.zip)

Not working "20250126 BTT rodent" using 24v PSU:

`.........................................................................................................................................`

Not working "20250126 BTT rodent" using 5v USB power:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1820
ho 0 tail 12 room 4
load:0x40078000,len:12844
load:0x40080400,len:3020
entry 0x400805e4
error:7
main/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
[MSG:Warnin
GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:Warning: Could not communicate with stepper driver!]
[MSG:Warning: Failed to initialize SD card]

```

It seems to boot when using 5v power from the USB port only, but when I use the 24v PSU the board will not boot. When using the older firmware from 14/12-24 the board works with the 24v PSU.

Does any of this make sense to you?

terjeio, 2025-01-28:
> Does any of this make sense to you?

Kind of, I believe I have found the error and comitted a [fix](https://github.com/grblHAL/ESP32/commit/ecc0fb43ee93904335ae423eab7874a57841e924). I have only tested with a breadboarded setup so you will have to try again.

enghamar, 2025-01-28:
Hi terjeio

Thank you for this update. I am looking forward to testing as soon as the web builder is updated to this build version.

enghamar, 2025-01-28:
Hi

After trying the newer firmware, I am not seeing any difference:

20250128 firmware with 5v usb power:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1820
ho 0 tail 12 room 4
load:0x40078000,len:12844
load:0x40080400,len:3020
entry 0x400805e4

GrblHAL 1.1f ['$' or '$HELP' for help]
```

20250128 firmware with 24v PSU:

`..............................................É!..þ.................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................ð..................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................À............................................................................................................................................................................................................................................................................................................................................................................ð....................................................................................................................................................................................................................................................................................................................ü.................................................................................................................................................................................................ÿ....................................................................................................................................................................................................................à...........................................................à..................................................ü....ÿ.........................................................................................................................................................................À.............................................................................................................................................ü...................................ø...................................................................À.............ÿ.......................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................ðü...........................................................................................................................................................ü.......................................................................................ü.....................................................À...........................................................................................à........................................................................................................................ÿ...............................................`

terjeio, 2025-01-28:
Odd that with 5V power it seems to work now - what output do you get from `$I` and `M122`?

terjeio, 2025-01-28:
When powered via 24V is the jumper removed?

[image]

enghamar, 2025-02-02:
Yes when the board is powered using 5V this jumper needs to be inserted, but I remove it before connecting the 24v PSU to mains power :)

[ESP32_BTT_RODENT.json](https://github.com/user-attachments/files/18632687/ESP32_BTT_RODENT.json)
[firmware (7).zip](https://github.com/user-attachments/files/18632686/firmware.7.zip)

Startup with 5v USB power:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1820
ho 0 tail 12 room 4
load:0x40078000,len:12844
load:0x40080400,len:3020
entry 0x400805e4
error:7
main/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
[MSG:Warnin
GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:Warning: Failed to initialize SD card]

```

M122 : 
```
ok
ok
ok
0 - Basic
ok
[TRINAMIC]
                      X       Y       Z      X2
Driver          TMC5160 TMC5160 TMC5160 TMC5160
Set current         500     500     500     500
RMS current         490     490     490     490
Peak current        692     692     692     692
Run current        9/31    9/31    9/31    9/31
Hold current       4/31    4/31    4/31    4/31
Global scaler   133/256 133/256 133/256 133/256
CS actual          4/31    4/31    4/31    4/31
PWM scale             5       5       5       5
vsense              N/A     N/A     N/A     N/A
stealthChop       false   false   false   false
msteps               16      16      16      16
tstep           1048575 1048575 1048575 1048575
pwm
threshold             0       0       0       0
[mm/s]                0       0       0       0
OT prewarn        false   false   false   false
OT prewarn has
been triggered    false   false   false   false
off time              5       5       5       5
blank time            1       1       1       1
hysteresis
-end                  5       5       5       5
-start                3       3       3       3
Stallguard thrs       0       0       0       0
DRIVER STATUS:
stallguard
sg_result             0       0       0       0
fsactive
stst                  *       *       *       *
olb
ola
s2gb
s2ga
otpw
ot
STATUS REGISTERS:
 X = 0x80:04:00:00
 Y = 0x80:04:00:00
 Z = 0x80:04:00:00
 X2 = 0x80:04:00:00
ok
```

$I : 
```
[VER:1.1f.20250128:]
[OPT:NMDSL2,100,1024,3,0]
[NEWOPT:ENUMS,RT+,ES,REBOOT,SED,RTC,WIFI,TMC=7,SD]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FLASH 4K]
[FREE MEMORY:168K]
[DRIVER:ESP32]
[DRIVER VERSION:250122]
[DRIVER OPTIONS:4.3.2]
[BOARD:BTT Rodent]
[AUX IO:0,2,0,0]
[WIFI MAC:8c:4f:00:11:e9:fc]
[IP:]
[PLUGIN:Trinamic v0.24]
[PLUGIN:WebUI v0.24]
[PLUGIN:SDCARD v1.20]
[PLUGIN:FS macro plugin v0.16]
ok
ok
ok
ok
ok
0 - Basic
ok
```

terjeio, 2025-02-02:
Looks almost ok - there is an issue with littlefs:

`main/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}`

To fix this erase the flash before flashing grblHAL.