grblHAL/core #668: Can’t change $73

mohamma393, 2025-01-29:
Hi
I have BLACKBOX32 controller 
Trying to to change $73=1 to $73=2 or $73=3
Tried with $70=0,1,2,3,4,5,6 and 7
Every time error52 show up
$73 only accept 0 or 1 
Tried with 3 different software 

[image]
[image]

How to deal with it to connect through AP please

terjeio, 2025-01-29:
> How to deal with it to connect through AP please

You have to compile and install a version that supports AP mode, you may use the [Web Builder](http://svn.io-engineering.com:8080/) for that.

FYI available options for $73 can be shown in the console by sending `$$=73`.

mohamma393, 2025-01-29:
Thank you so much, it’s working now

closed 2025-01-29 by mohamma393