grblHAL/core #669: Having issues Generating a working FW for Sienci SLB and Rapid Change ATC intergration

MattMead166, 2025-01-29:
Hello, 

I have a Sienci Labs Longmill Mk2.5 with SLB and a Rapid Change ATC that I am trying to integrate. I have been trying for almost a week straight now and have not had success. I have seen PATCH updates for 1/16, 1/18 and 1/28 which all added useful features but have NOT been able to correct my issues. I have attached the JSON files (and labeled them accordingly) I have used for different versions of FW. The main issues I am running into is: 

1) My "working 3-axis FW" only has THREE (3) axis available X,Y,Z but I have a current FW from Sienci with FOUR (4) axis available X,Y,Z,A as I have there rotary attachment as well as there Laser Attachment. 

2) my "NOT working 4-axis FW" has ALL Four (4) axis available when I build the FW X,Y,Z,A and ALL of the settings I need for MODVFD etc all looks good and is there but no matter which patch I have built from they ALL will NOT drive my second Y2 ganged motor even though I have that set to "ganged" and causes no movement/control of Y2 and skews the machine axis and renders it unusable.  I tried setting $8 for "Ganged axes dirrection invert" both ways also and the problem still persits. 

[image]

3) On **ALL** "working 3-axis" and "NOT working 4-axis" FW's no matter what options I choose in the plugin section of the web builder I do NOT get a FW that has Laser options for my machine. all of the $730-$743 "laser settings" are missing.  As well on ALL versions I am missing setting $668 to invert the TLS pin which is NOT allowing the IR sensor on my ATC to report correctly. 

[image]

[image]

4) On **ALL** versions I am missing setting $668 to invert the TLS pin which is NOT allowing the IR sensor on my ATC to report correctly.  I have added the "probing-add on" in the 3rd party plugins which has given me setting $459 that I have tried every combo I can think of. I can either get the "touch probe" to work and NOT the "IR sensor for tool detection" or I can get the "IR sensor for tool detection" to work correctly but then the "probe" does NOT work as it says it is in the "wrong state" now. It also gave me settings $457-$458 for PROBE aux input and TOOL PROBE aux input which I believe I have set correctly? Based off of the SLB board map you have attached to the web builder. 

[image]

[image]

[image]

I am hoping that you may be able to help me resolve my issues and get a working FW for my machine. Thank you very much in advance for taking the time to read over my issues and and help or guidance you can provide is very much appreciated. 

[MOTORS_NOT_WORKING - 3_AXIS_MATT TEST 3_partially working_NO ROTARY_NO LASER_firmware_STM32F4xx_LONGBOARD32.json](https://github.com/user-attachments/files/18594063/MOTORS_NOT_WORKING.-.3_AXIS_MATT.TEST.3_partially.working_NO.ROTARY_NO.LASER_firmware_STM32F4xx_LONGBOARD32.json)
[MOTORS_NOT_WORKING - 4_AXIS_MATT TEST 1_partially working_ADD ROTARY_ADD LASER\[LASER DOESNT SHOW\]_firmware_STM32F4xx_LONGBOARD32.json](https://github.com/user-attachments/files/18594062/MOTORS_NOT_WORKING.-.4_AXIS_MATT.TEST.1_partially.working_ADD.ROTARY_ADD.LASER.LASER.DOESNT.SHOW._firmware_STM32F4xx_LONGBOARD32.json)
[MOTORS_NOT_WORKING - 4_AXIS_MATT TEST 2_partially working_ADD ROTARY_ADD LASER\[LASER DOESNT SHOW\]_firmware_STM32F4xx_LONGBOARD32.json](https://github.com/user-attachments/files/18594061/MOTORS_NOT_WORKING.-.4_AXIS_MATT.TEST.2_partially.working_ADD.ROTARY_ADD.LASER.LASER.DOESNT.SHOW._firmware_STM32F4xx_LONGBOARD32.json)
[MOTORS_WORKING - 3_AXIS_partially working_NO ROTARY_NO LASER FW.json](https://github.com/user-attachments/files/18594064/MOTORS_WORKING.-.3_AXIS_partially.working_NO.ROTARY_NO.LASER.FW.json)

[MOTORS_NOT_WORKING - 3_AXIS_MATT TEST 3_partially working_NO ROTARY_NO LASER_firmware_STM32F4xx_LONGBOARD32.json](https://github.com/user-attachments/files/18594066/MOTORS_NOT_WORKING.-.3_AXIS_MATT.TEST.3_partially.working_NO.ROTARY_NO.LASER_firmware_STM32F4xx_LONGBOARD32.json)
[MOTORS_NOT_WORKING - 4_AXIS_MATT TEST 1_partially working_ADD ROTARY_ADD LASER\[LASER DOESNT SHOW\]_firmware_STM32F4xx_LONGBOARD32.json](https://github.com/user-attachments/files/18594067/MOTORS_NOT_WORKING.-.4_AXIS_MATT.TEST.1_partially.working_ADD.ROTARY_ADD.LASER.LASER.DOESNT.SHOW._firmware_STM32F4xx_LONGBOARD32.json)
[MOTORS_NOT_WORKING - 4_AXIS_MATT TEST 2_partially working_ADD ROTARY_ADD LASER\[LASER DOESNT SHOW\]_firmware_STM32F4xx_LONGBOARD32.json](https://github.com/user-attachments/files/18594065/MOTORS_NOT_WORKING.-.4_AXIS_MATT.TEST.2_partially.working_ADD.ROTARY_ADD.LASER.LASER.DOESNT.SHOW._firmware_STM32F4xx_LONGBOARD32.json)
[MOTORS_WORKING - 3_AXIS_partially working_NO ROTARY_NO LASER FW.json](https://github.com/user-attachments/files/18594068/MOTORS_WORKING.-.3_AXIS_partially.working_NO.ROTARY_NO.LASER.FW.json)

terjeio, 2025-01-30:
$668 is an optional stepper driver setting. Sienci added some setting numbers to their fork without reserving them so those that clashes with the upstream repo is not available for the same use as defined in the fork.
FYI $6 is used for inverting probe inputs, the toolsetter input can be inverted by changing it:

[image]

Do not enable any of the probe plugins since those may interfere with the way probe inputs are handled by the SLB.

I am working on the axis configuration issue, will commit an update soon if anything needs to change.

terjeio, 2025-01-30:
You may try a new build in the Web Builder now - testing is ongoing so the changes will not be committed until verified.

MattMead166, 2025-01-31:
Thank you for the updates Terje. I have spoke with Chris at Sienci also and he gave me a starting point for a new FW. I built it last night and tested it. I did notice (and Chris said he informed you also) that I am NOT able to set a second spindle to PWM2 for laser use. is makes the FW go crazy and then removes ALL $values for A axis. Additionally it seems that $180-$183 and $190-$193 are missing for HOMING feed rate and seek rate. other then that though this is a WORKING version of FW and I was able to use all functions on the Rapid Change ATC so thank you very much for that.

terjeio, 2025-01-31:
> I am NOT able to set a second spindle to PWM2 for laser use.

To enable PWM2 for laser use when you have two PWM spindles you have to set `$32=1` and disable laser mode for the standard PWM spindle by a `$9` flag:

[image]

Leave `$32` permanently set to 1, laser mode is turned off/on depending on which spindle is active.

> is makes the FW go crazy and then removes ALL $values for A axis

Not sure what you mean by this...
There is an issue with enabling A-axis when a ganged/auto-squared axis is also enabled, `$338` may be set to 15 when it should be 7. I'll look into this.

> Additionally it seems that $180-$183 and $190-$193 are missing for HOMING feed rate and seek rate.

These are gone from the Trinamic driver and [made optional](https://github.com/grblHAL/core/blob/master/changelog.md#20241208), enabled by a flag in the `$22` setting: _Per axis feedrates_.

MattMead166, 2025-02-05:
Hello Terje, Apologies for the late reply, I was out of town for work. For the laser, I do NOT have two PWM spindles. My Spindle is MODVFD and my laser is PWM. How would I go about setting that up then? As far as the FW going crazy. What I meant is I when I enable 4 axis and ganged y on the web builder it builds the FW as expected. But when I go to FW settings and set spindle 1 to MODVFD as it should be and then set spindle2 to PWM2 to be able to use my laser also when I reset the controller and then open FW again after enabling spindle2 then ALL A axis $ values are gone from the FW. I will look into $338 as you mentioned. I will also look into $22 for the HOMING values that were removed from the Trinamic driver. Sent from my iPhone

image.png (view on web)

terjeio, 2025-02-06:
>  My Spindle is MODVFD and my laser is PWM. How would I go about setting that up then?

Enable laser mode permanently with $31=1. It will then be turned on/off based on the active spindle.
And do no select PWM2 for the laser, select PWM when building.

> when I reset the controller and then open FW again after enabling spindle2 then ALL A axis $ values are gone from the FW. 

So you do not see them any more or are they reset to default?
When testing with ioSender I am not able to replicate this - I even see Trinamic driver settings for A which I should not.