grblHAL/core #678: MPG communication

MrKapim, 2025-02-09:
Hi @terjeio . Give me a help. I'm trying communicate by MPG but the answer is just "trash". Take a look on pics... 
This first is after reset the controller (STM743)

[image]

This other is after connect the IOSender

[image]

Here more info about my system to help you analyse...

[image]

[image]

MrKapim, 2025-02-09:
What do you think about what's happening? Did I configure something wrong?

terjeio, 2025-02-09:
Looks like incorrect baud rate to me. @dresco can help?

MrKapim, 2025-02-10:
Yes. I thinked the same, baudrate. But is correct 115200. I don't talk before but I monitoring all time, ff I reset the controller I don't receive nathen but if I connect the IoSender, I receive this "answers"... Just after connect USB, I mean.

terjeio, 2025-02-10:
> But is correct 115200.

Have you checked that with a scope?

dresco, 2025-02-10:
Agree it looks like incorrect baud rate, but not sure why?

Have just verified on my SKR3, and MPG output (115200 baud) is as expected on reset and real-time command. Have also used that UART for MODBUS testing with baud rates as expected.

```
GrblHAL 1.1f ['$' or '$HELP' for help]
<Idle|MPos:0.000,0.000,0.000|Bf:100,1024|FS:0,0|Pn:XZ|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|Bf:100,1024|FS:0,0|Pn:XZ>
```

terjeio, 2025-02-10:
Incorrect terminal configuration? Parity? Stop bits?

MrKapim, 2025-02-11:
Okay guys, I changed to others serials and only port 2 work. Port 3 don't communicate and port 1 have the result comment before. About configuration (baudrate, parity, stopbits) is the same for all 3 serials

dresco, 2025-02-11:
Afraid I'm not able to replicate here, can you share whatever configurations are necessary to recreate your build?

MrKapim, 2025-02-15:
Hi @dresco. I made several changes to test the ports and pins. Sets B14, B15 and D8, D9 do not work. Sets B10, B11 and A2, A3 work perfectly. But the A9, A10 set does not work, it communicates but the information is not correct (first image I sent). It is visibly clear that there is an error in baudrate and frequency. My configurations is standard, default...

dresco, 2025-02-16:
Okay thanks, will re-check here.

And just to confirm, you are defining each of these with SERIAL_PORT rather than SERIAL1_PORT or SERIAL2_PORT (helps me know where to look)..

dresco, 2025-02-16:
Am still unable to replicate here, on any ports I've tried. Baud rate is always 115200 as expected..

On the SKR3 map, I have tested MPG output on both the standard port definitions;
```
#define SERIAL_PORT    1   // TFT header, GPIOA: TX = 9, RX = 10
#define SERIAL1_PORT   32  // ESP-32,     GPIOD: TX = 8, RX = 9
```

And also tried a different pinout available on the ESP header for testing;
`#define SERIAL_PORT     12  // ESP-32,     GPIOB: TX = 14, RX = 15`

Maybe you could share your full build configuration - build options, map file etc?

dresco, 2025-02-16:
> Other detail... For the pins A9, A10 the serial.c use a HAL_RCC_GetPCLK1Freq() and for others use HAL_RCC_GetPCLK2Freq(). This definition is in serial.c.... Maybe a error on this hardware setup?

I don't follow? It uses peripheral clock 2 for USART1 & USART6, and peripheral clock 1 for USART2 & USART3. (Though have just noticed I do need to extend that check for the 61 pinout definition).

```
#if SERIAL_PORT == 1 || (SERIAL_PORT >= 10 && SERIAL_PORT < 19) || SERIAL_PORT == 6
#define UART0_CLK HAL_RCC_GetPCLK2Freq()
#else
#define UART0_CLK HAL_RCC_GetPCLK1Freq()
#endif
```

MrKapim, 2025-02-16:
Hi @dresco, thanks for effort in helping me. To make things easier and stay on the same page, I downloaded a new H743 driver and changed the following... 1. I deleted the definitions for other MB in platformio, leaving only my MB " [env:btt_skr_30_h743_tmc5160_bl128] " to save compilation time.           2. I changed the inputs for control from A5,6,7 to E5,6,7 in the map (where they are connected on my board).             3. I added serial2 to the map to test. The result is the same, it works for all ports as long as it is not MPG STREAM 0, that is, pins A9, 10. Only these pins are defined as serial 1, USART 1 and therefore STREAM 0. My fear is not being able to use this stream, port. Because I need 3 serials...

MrKapim, 2025-02-16:
Other detail, receive wrong datas just on A9, 10 with STREAM 0 selected, And after connect IOsender (by USB), this datas repeat continuous

MrKapim, 2025-02-16:
Of course to tests, I changed the MPG STREAM to 1 and 2 to select the other serial ports (and set this ports to 1, to select pins A9, 10)

terjeio, 2025-02-16:
Check that the correct stream is assigned with the $PINS command.

dresco, 2025-02-16:
Looking closer, I'm not sure how it can even compile with the port definitions in your earlier screenshots?

PA9/PA10 are on USART1 (serial port definition 1)
PB10/PB11 are on USART3 (serial port definition 3)
PD8/PD9 are also on USART3  (serial port definition 32)

MrKapim, 2025-02-16:
I'll try answer the 2 guy's...  All time I check on IOsender the stream selected to check the pins selected (pins A9, 10). If any serial selected with number 1 gave me the error. The other ports or pins is working. I'm not "mixing" the definitions. If I select a port, I have to select its number according to the definitions in serial.c, so that I don't call the same port or stream twice. But Okay, I'm select serial_port = 1 and MPG stream = 0 (pins A9, 10), serial1_port = 2 and modbus stream = 1 (pins B10, 11) and serial2_port = 3 and keypad stream = 2 (pins D8, 9). This is my first configuration and the MPG connection show me the error describe above in first pic

dresco, 2025-02-17:
Okay, am getting somewhere now. Have put the BTT bootloader back on my board, and appears it is affecting the USART1 clock setup, will have to look into that..

Note however that the serial ports in your last post don't match the pin numbers you are describing - or the screenshot in your first post?

serial port definition 1 is for USART1 on pins A9/A10 (on the TFT header)
serial port definition 2 is for USART2 on pins A2/A3 (on the TH0/TH1 thermistor inputs)
serial port definition 3 is for USART3 on pins B10/B11 (on the WiFi header)

(Btw, will the thermistor input pins even work for this purpose? I haven't looked at the schematic in detail)..

dresco, 2025-02-17:
@MrKapim Have just pushed an update that I think should solve the USART1 issue when started from bootloader. Please test and let me know, thanks!

dresco, 2025-02-18:
> There is a typo on line 154 of serial.c

Thanks fix committed..

> I deleted the definitions for other MB in platformio, leaving only my MB " [env:btt_skr_30_h743_tmc5160_bl128] " to save compilation time.

Btw, I skipped over this earlier, but you don't need to build all environments. If you click on the bottom bar here, you can choose which env you are building;

[image]

MrKapim, 2025-02-24:
Hi @dresco . Question for you... This change that you made and allowed serial 1 (Stream 0) to work correctly (remembering that I am using all 3 serial ports) could cause some excessive consumption on the hardware???  Because my MB broke down (whenever I turn it on (powering the 12Vdc), the "power" LED comes on and goes off" showing that the voltage is dropping, that is, there is a short. Another detail is that the controller (H743) is "frying, boiling". I can't put my finger on it.  Remember...  I didn't change anything else, whether in the hardware or firmware after your changes, except the pins 5, 6, 7 de GPIOA para GPIOE and erased the platformio about the compilations for others MB.

MrKapim, 2025-02-25:
Hi @dresco . Forget the question and forgive me about this. I found the error, problem. But if you block or not enable the pins A2, A3 for communication maybe is better (in serial.c). Check the pics...  The circuit for thermistors have a Vref (VDDA name) and this use the controller ref pin. This ref is 3.3Vdc and the communication use 5Vdc (in my case).

[image]

[image]

dresco, 2025-02-25:
> I found the error, problem. 

Maybe you can find another pinout that will work if you need 3 serial on this board?

Have just double checked against the datasheet & they are all already listed in serial.c except for USART10. However, that only has one usable pinout on the 100pin ICs, and is no good for this board as already in use for Z step &direction (PE2/PE3).

> But if you block or not enable the pins A2, A3 for communication maybe is better (in serial.c). 

Can't block this as it's a valid serial pinout used on other boards. As always, check your schematic carefully if adding additional inputs/outputs to the board map file..

dresco, 2025-04-17:
> Hi [@dresco](https://github.com/dresco) , I'm here again about the serial communication and MPG mode.

Hi, I'll take another look when I can. Am travelling just now, so may be a couple of weeks before I get the chance to test properly with hardware..

MrKapim, 2025-06-07:
Hi @dresco, Did you enjoy the retreat? I hope you had fun. I also took a break from my project and am now getting back to it. Were you able to analyze anything? Just a reminder... The serials send correctly but do not receive or accept commands... In other words, I am receiving the GRBL status but for example when I send "$H" with MPG and KEYPAD enabled, nothing happens. Again, only USART3 works, the others do not (pins B10 B11, D8 D9)

closed 2026-02-27 by terjeio