grblHAL/core #682: 2 axis homing

Bartekn86, 2025-02-15:
Hi,  teensy 4.1, imxrt1062
Could you add to webbuilder changes thats allow homing only 2 axis ? x and y  please ? 
Could you decrees number of axis to 2 ?

terjeio, 2025-02-15:
> Could you add to webbuilder changes thats allow homing only 2 axis ? x and y please ?

You configure that run time by settings $44 - $46:

[image]

> Could you decrees number of axis to 2 ?

Can but don't want to because that will be a lot of work. And I guess it would break most if not all senders.

Bartekn86, 2025-02-15:
thanks for your time, could you tell me please, what software is on the screen ?

closed 2025-02-22 by Bartekn86