grblHAL/core #685: Welcome Message

StefanNetzer, 2025-02-22:
Hi, from grbl I am used to get a welcome message. So when I uploaded the software to the Bigtreetech I expected such a message as well. Finally I tried a $$ which reported back as expected. There is function output_welcome_message in grbllib.c. 

I have seen a request in a Lightburn forum indicating that this missing prompt may cause an issue. https://forum.lightburnsoftware.com/t/connecting-to-grblhal-without-welcome-message/92961

terjeio, 2025-02-22:
> I expected such a message as well.

[You cannot](https://github.com/grblHAL/core/wiki/For-sender-developers). For some connections it is impossible for the controller to know when a client has connected. Since you have not included any information about the board, the grblHAL version and the connection metod I do not have a better answer for now.

StefanNetzer, 2025-02-22:
The compile options in the Webbuilder are:

{
  "driver": "RP2040",
  "URL": "https://github.com/grblHAL/RP2040",
  "board": "BOARD_BTT_SKR_PICO_10",
  "symbols": [
    "OVERRIDE_MY_MACHINE=1",
    "USB_SERIAL_CDC=1",
    "PROBE_ENABLE=1",
    "TRINAMIC_ENABLE=2209",
    "TRINAMIC_R_SENSE=110",
    "TRINAMIC_UART_ENABLE=1",
    "LB_CLUSTERS_ENABLE=1",
    "ESTOP_ENABLE=1",
    "COREXY=1",
    "SPINDLE0_ENABLE=11",
    "N_SPINDLE=1"
  ],
  "docker_instance": "/RP2040",
  "grblHAL": "webBuilder"
}

terjeio, 2025-02-22:
My version of LightBurn connects, this with a Pi Pico running the firmware:

[image]

You have an older version that does not assert the DTR line as it should?

StefanNetzer, 2025-02-22:
I have not installed Lightburn yet, so I tested with the Arduino IDE. Let me install Lightburn and I will report back. Thank you so much for the instant help 👍

closed 2025-02-24 by terjeio