grblHAL/core #688: Jog improvements possible?

ipsod, 2025-02-24:
Hi,

Would it be in-scope to add realtime control over speed, accel, and position target, for use with robotics, where planning happens on a PC and motion is sometimes corrected by it in real time, and for manual remote control of machines?  You can get the realtime control with different libraries, like FastAccelStepper, but then you're left to do coordinated motion control on your own.

GRBL 1.1-style jogging's limitations are mentioned in its [docs](https://github.com/gnea/grbl/blob/master/doc/markdown/jogging.md).  To summarize: because of the motion buffer, there's always a fairly significant delay between control and response.  If you could get it down to like, a predictable 5 or 10 ms (or better), with a lot easier control mechanism, I think it'd make a lot of cool projects a lot easier.

terjeio, 2025-02-25:
> Does it look that way for you, too?

Yes - jogging can be more or less complicated depending on which device controls it. Input from a joystick has to be converted to a pulse train to control the steppers and I assume it will be for least two axes so coordinated motion is likely needed.

[stepper2.c](https://github.com/grblHAL/core/blob/master/stepper2.c) is an example of how I use step injection to control a stepper directly. This is somewhat different from jogging since motion has to be in parallell with normal motion but anyway shows how a motor can be controlled without going through the planner.

terjeio, 2025-03-01:
> Do you know of any software that would be more suitable?

Not really.

> ... just wondered if it might be expanded to cover more classes of machines.

It can, new [kinematics](https://github.com/grblHAL/core/tree/master/kinematics) can be added. An there is core support for up to 8 axes. I have provided that I believe is a very flexible framework that others can add to, and I do not have unlimited time (or money) available for adding stuff that might be of use. It would be great if somebody else would take on tasks like this - I would happily support them.

ipsod, 2025-03-01:
Sorry, I didn't mean to complain about grblHAL.  I'm trying to solve an actual problem I'm having that makes it not work as well as LinuxCNC for a type of machine I run.  Adding kinematics doesn't fix the issue with slow response time for manual control or to adjust paths to realtime inputs.

Yeah, I totally understand.  I don't see myself taking the time to do it, either.  I mean, I'd like to, but, realistically, it's not on the horizon.

Thanks for your work.  I definitely find it useful and really cool as-is.  It just doesn't work for my current project, is all.  But, I'm working on a Python client for grbl-style machines that I'm pretty excited about.  Should make it easy to build applications that interface with grbl-driven machines.  I just really wish it could reach into realtime, for the machine I'm working on, and for some I've got in mind.