grblHAL/core #693: Request to add T0 as an "empty" tool slot.

LoganFraser, 2025-02-27:
As shown below, this is a request for Tool 0 (T0) to be considered an empty tool for unloading a toolchanger.  This is common to Linuxcnc and FluidNC that I'm aware, of, and possibly others.

The current firmware appear to just ignore T0 requests.  

[image]

terjeio, 2025-02-27:
You mean this?

[image]

Added as an option recently.

LoganFraser, 2025-03-02:
aha!  Thank you!

closed 2025-03-02 by LoganFraser

terjeio, 2025-03-03:
```
o300 if [[#<_vminor> ge 241025] and [PRM[675, 0] eq 1]]
(print, Supports T0M6)
o300 else
(print, Do not support T0M6>)
o300 endif
```