grblHAL/core #699: X-Axis Drifting After Updating to grblHal Beta 1.0.1.3

WhatsUGonnaDo, 2025-03-04:
Hi, 
I've run into a serious issue with X-position drift after updating my firmware from the stable grblHal Release 0.9.9.8 to the beta Release 1.0.1.3. The working coordinates are shifting unexpectedly, and by the end of a machining operation, the X position is completely off—even though no alarms are triggered.

What I’ve Checked & Tested:

 No loose couplings, slacks, or improperly tightened BK/BF blocks.
 No backlash compensation is enabled.
 Using Closed Loop CL57T V4.1 drivers from StepperOnline, which have been working fine so far.
 Adjusted parameters like pulse width and pulse delay—no real difference.
 Upgraded to the latest ioSender version 2.0.45, which resolved jog command issues.

Confirming It’s a Firmware Issue:

Downgraded back to the stable 0.9.9.8 → Machined successfully with no issues.

Upgraded again to 1.0.1.3 → Identical X-axis drift problem.

Additional Notes:
The toolpath is a challenging one—a large adaptive operation with many small movements, running at 2800mm/min—but this was never an issue with the older stable release.
I've also sent a Contour OP along the same geometry 

I’ll be attaching the G-code file if anyone wants to emulate it.

WhatsUGonnaDo, 2025-03-04:
[image]

WhatsUGonnaDo, 2025-03-04:
[image] The area in red helps visualize how much it drifted to the right

terjeio, 2025-03-04:
I have no idea what these version numbers relate to, are you confusing grblHAL with some other software?

WhatsUGonnaDo, 2025-03-05:
> I have no idea what these version numbers relate to, are you confusing grblHAL with some other software?

Yes my bad, sorry, those are Flexi-HAL firmware references, I meant the 20231229 build and the 20250223 respectively, being the issue on the latest.

Robin-SeaThrough, 2025-03-05:
> BTW is jerk acceleration enabled in that version?

As far as I can see, and from what Andrew Marles have said earlier, no, jerk acceleration isn't enabled in this version. I didn't see any jerk related settings in my machine. 

I'm running the same FlexiHAL 1.0.1.3 firmware at the moment (I wanted to get the new Core changes around Stop/Reset/TLR). That's the one with grblhal core 20250223. 
I also have a fairly similar PrintNC using almost the same stepper drivers (CL57T V4 instead of CL57T V4.1) 

While I haven't seen the issue reported here on my machine, I did talk to WhatsUGonnaDo while he was testing, and updating the firmware clearly made the difference. 

Not sure why it doesn't seem to be happening on my machine, maybe I just haven't run a program that is big enough with enough back-forth motions yet, I'll see if I can run his Gcode today. 

One thing that is different between his and my setup is that he has the uFlexiNET module installed (in fact I sold my module to him recently), which gives the FlexiHAL an ethernet connection and SD card reader. 
Iirc WhatsUGonnaDo did try both firmware versions with and without the ETH_SD functionality. 

I did see Andrew mention that there seems to be some issue related to SD card and ethernet with the 1.0.1.3 firmware, and that he thought that the drifting issue may be related to clock config and that it was fixed back in 2022. 
Maybe there is some regression somewhere causing this behaviour? 

I have no way of knowing if this is from the Core or if its specific to the FlexiHAL firmware.