grblHAL/core #716: Error 1 when trying to set a variable (e.g. `#1=1`)

axxel, 2025-03-22:
I'd like to use variables. [This comment](https://github.com/grblHAL/core/discussions/309#discussioncomment-6003124) from @terjeio made me believe they are supported.

According to my web research a simple macro like the following should work:
```
#1=1
g0 x#1
```
But it fails with `Error 1` in the first line. I feel like I'm making something obviously wrong but I simply can't find it.

karoria, 2025-03-23:
Numbered parameters range is 31 to 5000. Use within it. Read [this](https://github.com/grblHAL/core/wiki/Expressions-and-flow-control) page for more details.

axxel, 2025-03-24:
Thanks for trying to help but `#101=1` was actually my very first experiment. Also the `Error 1: (Expected command letter)` sounds more like the `#` is already where the parser bails, so the following number is not even parsed?

terjeio, 2025-03-24:
Is expression support enabled in the build? Check that `EXPR` is present in the `NEWOPT` list:
```
$i
[VER:1.1f.20250225:]
[OPT:VNMSL,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,EXPR,SED]
```
A test:
```
#1=10
#101=22
$#=1
[PARAM:1=10]
$#=101
[PARAM:101=22]
```

axxel, 2025-03-25:
> Is expression support enabled in the build?

Good question. I used the web-builder and I double checked all available options but could not find anything that seemed related to expression/parameter support.

terjeio, 2025-03-25:
Look near the bottom of the _Advanced features_ tab...

axxel, 2025-03-25:
> Look near the bottom of the _Advanced features_ tab...

Oh my. I've gone through all of that multiple times... 🫣. I guess my brain filtered that line due to the fact that it starts with "RS274". But since that is explicitly mentioned in the wiki page that @karoria already pointed to, I have zero excuses. Sorry for wasting all your time.

closed 2025-03-25 by axxel