grblHAL/core #720: Hold wont Continue sometimes on Lathe Mode

Reprosense, 2025-03-27:
Bug Happens when verbose output shows
<Hold:1|MPos:-183.748,0.000,-112.274|Bf:16,923|Ln:55|FS:1986,2500,4274>

the Hold will always on "hold pending"( HOLD:1) mode , and dont want to change into " Hold : 0 "  and my gcode wont continue after click play again , it should be stopped first and re-stream the gcode , i thinks its because "EXEC_CYLE_COMPLETE" wont be executed , i didnt get any other data than that , your help really appreciated , thanks in advance ,

Firmware : 

[VER:1.1f.20250317:]
[OPT:VNZHSL,35,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,NOPROBE,ES,LATHE,SS,SED,PID]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[FREE MEMORY:32K]
[DRIVER:STM32F446]
[DRIVER VERSION:230922]
[BOARD:Protoneer v3]
[AUX IO:1,2,0,0]
ok

Note : I test it with clicking hold & play within 0.5 second interval within the run of gcode , here i sent you the gcode 

[Trial Hold Bug.txt](https://github.com/user-attachments/files/19478059/Trial.Hold.Bug.txt)

terjeio, 2025-04-17:
Sorry for responding late.
I have commited some changes recently that may affect this - but not sure. Please try the latest version.
When testing I have managed to end up in IDLE mode mid program and then beeing unable to transition to CYCLE again. But this happens very rarely for me so hard to pin down which state/state transition that causes this so it may take time before I am able to resolve the issue. BTW I have not been able to end up in the HOLD:1 state yet.
Does this issue severly affect normal operation?

Reprosense, 2025-04-18:
hi terje , thanks for responding , i've been try the newest firmware and run it for 10x times now , it seems okay now