grblHAL/core #721: Specific working mode for CO2 laser on parameter $32!

Deltaplc, 2025-03-27:
Good morning
Sorry to bother you....
I don't know if I've discovered hot water.....but I certainly think I've thought of(and tested) a valid solution to solve the problem!
I will try to explain myself with some examples, to help you better understand the problem that occurs when engraving images.
I have a home-made CO2 laser derived from a K40.

Example of engraving a photo in any mode (Stucki, Jarvis, dithering, etc.) except grayscale.

K40 operation "original configuration":
Fragment of a hypothetical gcode:
G1 X2.5F6000 off laser
G1 X0.1 on laser
G1 X20.5 off laser
G1 X0.1 on laser
G1 X2.5 off laser
G1 .....
The potentiometer (manual analog) supplies a constant voltage to the IN pin (pwm) and the board turns the laser beam on and off via the L (enable) pin.
Example of this configuration:
The sniper has his eye on the scope, always aimed at the target, and he only has to pull the trigger to shoot!
Result:
The IN pin is always at the desired voltage and already keeps the power supply "capacitors charged", it just needs to fire!

Custom laser with any board and firmware derived from grbl and with lightburn or lasergrbl software etc...:
Fragment of a hypothetical gcode:
M4
M3
G1 X2.5F6000 S0
G1 X0.1 S200
G1 X20.5 S0
G1 X0.1 S200
G1 X2.5 S0
G1 .....
The voltage on the IN(pwm) pin, varies from zero to S200, and the L(enable) pin, turns the laser beam on and off, following the strategy that depends on the M3/M4.

Example of this configuration:
The sniper must re-center his eye in the crosshairs every time he has to shoot and then pull the trigger!
Result:
The IN pin does NOT have a constant voltage, but varies between zero volts and S200,
the power supply "capacitors are not constantly charged" and when it fires it is not ready to immediately parry at the desired power!
General problem:
Loss of much detail when using dithering, Jarvis or similar methods at high speeds!

My thoughts Changes to be made to the firmware:
On the solution to this problem:

$32= 0 = CNC; 1= Laser ; 2= ​​Lathe ; 3=Laser_Co2
SPINDLE_ENABLE_PIN = Behavior unchanged, behaves as it does now.
SPINDLE_PWM_PIN = M3/M4 S.....= S1 / S1000 pwm regulation, does NOT consider the S0 parameter for pwm regulation.
M3 / M4 = Turns off with M5 or M2 command. (Now, if $34 is greater than 1, M3 does not turn off if M5 or M2 command is given!)
$34 = Possibility to vary the value from an M(....)S(...) command. In order to adjust and refine the resolution.
Result (already tested):

The voltage on the IN(pwm) pin remains constant, because it does NOT consider the S 0 value as a power variation! If $34 is
slightly higher than the desired power, it improves further!
The laser beam will still turn on and off, using the enable pin (L pin) which will keep the same current working strategy.
When using the M5 or M2 commands, the laser beam is switched off and made safe!
It is also possible to refine all this using the function supported by the $34 command, extending the possibility of varying its parameters with M(....)(S....) commands.
I have already tested this type of configuration manually and using the $34 parameter...
The difference in the result on the finished product is truly abysmal!
At the moment it seems to me that no one has a custom configuration for the control of co2 lasers!
In my opinion it is fundamental since the behavior of the laser beam ignition is profoundly different from that of diode lasers!
I hope you can make these changes, because it will bring huge benefits to those who use grblHal to control their custom co2 laser!

I am available if you need to do tests or for anything about it!
Thank you in advance.
Regards

Deltaplc, 2025-03-27:
Attached are 2 images engraved with the same parameters (speed 340mm/s power 7.6%).
Number 1 is with the current operating strategies. Number 2 is with the simulation of the operating strategy that I would like you to bring to the firmware! Currently, if I wanted to have a result similar to image 2, the only solution would be to raise the power a lot! But the result would still be terrible and not even comparable!

[image]

OrsoLaser, 2025-03-28:
Hi
I wanted to add another option....
There is a function in ligtburn "use G0 movements for overscan" that could affect the pwm output, since in laser mode M3/M4 do not work with G0 movements!!!
It is likely that if you decide to make the changes I asked you will have to take this into account!
I send you a fragment of gcode in question:
G0 X110.626Y89.399
M3
; Layer C00
G91
G0 X8.333
G1 X0.1F20000S80
G1 X0.1S0
G1 X0.2S80
G1 X0.1S0
G1 X0.1S80
G1 X0.1S0
.....

G1 X0.1S80
G1 X0.1S0
G1 X0.1S80
G1 X0.1S0
G1 X0.1S80
G0 X8.334
G0 Y-0.1
G0 X-8.334
G1 X-1
G1 X-0.1S0
G1 X-0.2S80
G1 X-0.1S0
G1 X-0.2S80
G1 X-0.1S0
G1 X-0.3S80
G1 X-0.1S0

Deltaplc, 2025-03-28:
Hi
I wanted to add another option....
There is a function in ligtburn "use G0 movements for overscan" that could affect the pwm output, since in laser mode M3/M4 do not work with G0 movements!!!
It is likely that if you decide to make the changes I asked you will have to take this into account!
I send you a fragment of gcode in question:
G0 X110.626Y89.399
M3
; Layer C00
G91
G0 X8.333
G1 X0.1F20000S80
G1 X0.1S0
G1 X0.2S80
G1 X0.1S0
G1 X0.1S80
G1 X0.1S0
.......
G1 X0.1S80
G1 X0.1S0
G1 X0.1S80
G1 X0.1S0
G1 X0.1S80
G1 X0.1S0
G1 X0.1S80
G0 X8.334
G0 Y-0.1
G0 X-8.334
G1 X-1
G1 X-0.1S0
G1 X-0.2S80
G1 X-0.1S0
G1 X-0.2S80
G1 X-0.1S0
G1 X-0.3S80
G1 X-0.1S0

terjeio, 2025-03-29:
Do you have a DC or RF excited tube? And at what kind of resolution (DPI) at 340 mm/s?

If I understand your proposal you want laser on/off to be controlled by distance in G1 mode, kind of like [PPI mode](https://github.com/grblHAL/Plugins_laser/#laser-ppi). If less/equal to some distance then turn laser on else turn it off?

Deltaplc, 2025-03-29:
> Hai un tubo eccitato DC o RF? E a che tipo di risoluzione (DPI) a 340 mm/s?

Hello
It's the classic glass tube with its original k40 power supply....

Deltaplc, 2025-03-29:
> Se ho capito bene la tua proposta, vuoi che l'accensione/spegnimento del laser sia controllato dalla distanza in modalità G1, un po' come [la modalità PPI](https://github.com/grblHAL/Plugins_laser/#laser-ppi) . Se è inferiore/uguale a una certa distanza, allora accendi il laser, altrimenti spegnilo?
No no, it was just an addition to the main request! That is fundamental!

Deltaplc, 2025-03-29:
> Se ho capito bene la tua proposta, vuoi che l'accensione/spegnimento del laser sia controllato dalla distanza in modalità G1, un po' come [la modalità [...]

No no, it was just an addition to the main request! That is fundamental!

terjeio, 2025-03-29:
You want S0 to turn off the laser while keeping the PWM output as-is?
If so it seems I have implemeted the spindle code incorrectly and a simple `else` will correct it. That is with _RPM controls spindle enable signal_ enabled in the settings:

[image]

Which driver are you using?

Deltaplc, 2025-03-30:
Good morning
At the moment I'm using an Arduino Due.(SAM3X8E)
But I'm also testing a board with esp32.....

Deltaplc, 2025-03-30:
> Vuoi che S0 spenga il laser mantenendo l'uscita PWM così com'è?

Yes exactly, you should only deactivate the enable output, without modifying the PWM!

> Se è così, sembra che abbia implementato il codice del mandrino in modo errato e un semplice `else`lo correggerà. Questo con _il segnale di abilitazione del mandrino dei controlli RPM_ abilitato nelle impostazioni:
> 
> [image]

I tried to see if one of these options would work for me, but at the moment none of them do what I need. 
There are these options: 0-1-3-5-7 ? There are no others, right?

terjeio, 2025-03-30:
$9=3 or $9=7 is what you need - $7, And $32=0 for now. I have modified the Web Builder code so you have to flash a fresh version, if you build locally you have to change a function.
FYI M5 will currently not turn the PWM off - more changes has to be made for that.

OrsoLaser, 2025-03-30:
I cloned :
git clone --recurse-submodules https://github.com/grblHAL/SAM3X8E.git

I have successfully compiled and downloaded with arduino ide....
I tried with $32=0 and $9 = 3 or $9=7......But none of them do what I need! The pwm does not stay stable at the same level when the enable output changes to on/off.....

> - $7,

I don't have the $7 parameter!? 
Am I doing something wrong?

Deltaplc, 2025-03-30:
Ho clonato:
git clone --recurse-submodules https://github.com/grblHAL/SAM3X8E.git

Ho compilato e scaricato con successo con arduino ide...
Ho provato con $32=0 e $9 = 3 o $9=7...... Ma nessuno di loro fa quello di cui ho bisogno! Il pwm non rimane stabile allo stesso livello quando l'uscita enable cambia in on/off.....

7 dollari,
Non ho il parametro $7!?
Sto sbagliando qualcosa?

terjeio, 2025-03-30:
> git clone --recurse-submodules https://github.com/grblHAL/SAM3X8E.git

I have updated the Web Builder, _not_ github. Replace this function in _driver.c_ if compiling locally:

```
static void spindleSetSpeed (spindle_ptrs_t *spindle, uint_fast16_t pwm_value)
{
#ifdef SPINDLE_PWM_CHANNEL
    if (pwm_value == spindle->context.pwm->off_value) {
        pwmEnabled = false;
        if(spindle->context.pwm->settings->flags.enable_rpm_controlled) {
            if(spindle->context.pwm->flags.cloned)
                spindle_dir(false);
            else
                spindle_off();
        } else if(spindle->context.pwm->flags.always_on) {
            if(PWM->PWM_SR & (1 << SPINDLE_PWM_CHANNEL))
                PWM->PWM_CH_NUM[SPINDLE_PWM_CHANNEL].PWM_CDTYUPD = spindle->context.pwm->off_value;
            else {
                PWM->PWM_CH_NUM[SPINDLE_PWM_CHANNEL].PWM_CDTY = spindle->context.pwm->off_value;
                REG_PWM_ENA = (1 << SPINDLE_PWM_CHANNEL);
            }
        } else
            REG_PWM_DIS = (1 << SPINDLE_PWM_CHANNEL);
    } else {
        if(PWM->PWM_SR & (1 << SPINDLE_PWM_CHANNEL))
            PWM->PWM_CH_NUM[SPINDLE_PWM_CHANNEL].PWM_CDTYUPD = pwm_value;
        else {
            if(spindle->context.pwm->flags.cloned)
                spindle_dir(true);
            else
                spindle_on();
            pwmEnabled = true;
            PWM->PWM_CH_NUM[SPINDLE_PWM_CHANNEL].PWM_CDTY = pwm_value;
            REG_PWM_ENA = (1 << SPINDLE_PWM_CHANNEL);
        }
    }
#else
    if (pwm_value == spindle->context.pwm->off_value) {
        pwmEnabled = false;
        if(spindle->context.pwm->settings->flags.enable_rpm_controlled) {
            if(spindle->context.pwm->flags.cloned)
                spindle_dir(false);
            else
                spindle_off();
        } else if(spindle->context.pwm->flags.always_on) {
            SPINDLE_PWM_TIMER.TC_RA = spindle->context.pwm->period - spindle->context.pwm->off_value;
            SPINDLE_PWM_TIMER.TC_CMR &= ~TC_CMR_CPCSTOP;
            SPINDLE_PWM_TIMER.TC_CCR = TC_CCR_CLKEN|TC_CCR_SWTRG;
        } else
            SPINDLE_PWM_TIMER.TC_CMR |= TC_CMR_CPCSTOP; // Ensure output is low, by setting timer to stop at TCC match
    } else {
        SPINDLE_PWM_TIMER.TC_RA = spindle->context.pwm->period == pwm_value ? 1 : spindle->context.pwm->period - pwm_value;
        if(!pwmEnabled) {
            if(spindle->context.pwm->flags.cloned)
                spindle_dir(true);
            else
                spindle_on();
            pwmEnabled = true;
            SPINDLE_PWM_TIMER.TC_CMR &= ~TC_CMR_CPCSTOP;
            SPINDLE_PWM_TIMER.TC_CCR = TC_CCR_CLKEN|TC_CCR_SWTRG;
        }
    }
#endif
}
```

I'll update github if this works as intended - _after_ changing all the other drivers to match. And likely with further changes.

And please post comments in english, if not I will respond in burmese or something like that when I can be bothered...

Deltaplc, 2025-03-30:
> And please post comments in english, if not I will respond in burmese or something like that when I can be bothered...

Sorry!!! In my hurry I forgot to translate it!!
Now I'll try and let you know!

Deltaplc, 2025-03-30:
I have done some field tests with my engraver.....
With:
$32=0
$9=3 / $9=7  (Nothing seems to change between the two configurations 3 or 7)
The machine keeps the pwm constant and works with the enable pin on/off as I would like!!! But the machine goes very slow and if it receives G0 commands it does not turn off the laser.

With:
$32=1
$9=3 / $9=7 (Nothing seems to change between the two configurations 3 or 7)
The machine keeps the pwm constant and works with the enable pin on/off as I would like!!! It seems to work fine with the image engraving speeds as before. And it also works fine with G0 commands by turning off the enable pin.
Even with the LB_CLUSTERS mode enabled everything seems ok.

With the new change, in any case, the $34 parameter does NOT work anymore!! Changing its value does NOT make any changes!!
As you said, the M5 command does not turn off the PWM at the end of the processing.

So, with $32=1 and $9=3 it seems to have worked!
 If you can get the $34 function to work as well, that would be awesome....

terjeio, 2025-04-02:
Replace _driver.c_ with this version, with luck it may work:

[driver.zip](https://github.com/user-attachments/files/19571226/driver.zip)

> There is a function in ligtburn "use G0 movements for overscan"

Overscan should be G1S0 motion with the same feedrate as engraving to ensure the PWM output does not change on the G0/G1 transition when laser mode is enabled. And the overscan distance should be large enough to let the PWM output to go to the programmed rate (at the end of acceleration).

> With the new change, in any case, the $34 parameter does NOT work anymore!! 

Perhaps the attached version does, if not how do you expect it should work?

terjeio, 2025-04-02:
M5 should at least work now.

> The $34 parameter seems NOT to work!!

What do you expect from this setting and with which setting value?

Deltaplc, 2025-04-02:
> M5 should at least work now.
I'm sorry but I haven't been able to do thorough testing to give you detailed information...I've only bench tested. I don't want to give you incorrect information....

Yes you are right, now M5 works!

> What do you expect from this setting and with which setting value?

It seems like $34 works but not in the right way....now the value $34 works at the end of the cycle after the M5 command has been received.
The pwm should load the value set by $34 every time the S0 value is read and the laser is turned off by the enable output....basically it should work as it did before the modification!

terjeio, 2025-04-02:
> The pwm should load the value set by $34 every time the S0 value is read and the laser is turned off by the enable output....basically it should work as it did before the modification!

I do not agree, at least when used with a CO2 laser power supply that needs time to settle. The PWM output should remain constant to allow a well defined leading edge. Or do you need the leading edge to have a slope? If so what about the trailing edge?
And do you know what the response time is for your power supply - e.g. from 0 to 100% PWM?

Deltaplc, 2025-04-02:
When I did some tests on image engravings with the jarvis-dithering-etc method, not on simple engravings... I noticed that using the $34 parameter with a slightly higher power than the one set for engraving (constant) increases the definition of the image details even more!
If you manage to make the $34 work as before the modification, I will try to make two images, with and without the parameter activated, I will show them to you and then you can see the difference...

Deltaplc, 2025-04-02:
All the modification I ask you is to be able to improve all the details in the engravings of the images (photos), for the engravings of writings or simple figures, there is no need to use the parameter $34, for this reason at the beginning I asked you about the possibility of setting it with an M parameter....
To be able to manage it more easily and also modify it via gcode...

terjeio, 2025-04-02:
>  I noticed that using the $34 parameter with a slightly higher power than the one set for engraving (constant) increases the definition of the image details even more!

Ok, I see - thanks for explaining why. Using $34 to control such a feature is not a good idea - a M-code (which one?) or a new setting would be better. This is not something I can do by modifying a single file - I would have to add a core call to set a value, which IMO should be relative to the programmed S rate - a percentage? And a plugin for the M-code. And perhaps also a new $9 option should be added to enable it...

Deltaplc, 2025-04-02:
Let's take it one step at a time, let's see that the changes we are making work and lead to improvements on the engravings!
If it's too complex to transform the $34 into an M parameter, forget it, it's not necessary!

> And a plugin for the M-code. And perhaps also a new $9 option should be added to enable it...

Probably the only essential thing is to add a value to $9 (or some other parameter) to activate and deactivate the configuration we are creating, because not everyone needs it....
I am always available for all the tests you need!!

Deltaplc, 2025-04-03:
> which IMO should be relative to the programmed S rate - a percentage?

If I understand what you mean, I had thought about it too.... create a parameter similar to $34 that takes the S value of the engraving + $xxx as a reference.
Example: engraving value (constant pwm) S80 + $(new $34) = 5% result: when the S0 command arrives the pwm rises and takes the value 84....
This would make this engraving strategy much easier, because you would no longer need to manually raise the $34 value based on the power set for image engraving!

Deltaplc, 2025-04-03:
[image]

Deltaplc, 2025-04-03:
There is a problem with the new modification!!
Continuing to do some tests, it seems that the enable output does not activate until inside the engraving it is given an S0 for the first time!!
I'll give you an example: engraving of Hello, it starts to engrave the letter H and does not engrave it, then it moves on to all the others and engraves them all correctly!!
I didn't notice before because I had only made "full" incisions, and not a simple circle perimeter incision or cut!

Deltaplc, 2025-04-03:
Resetting the board, the problem disappears, at the first engravings.... It seems to be triggered by the change of power between a single engraving and the other!
I'll try to explain: after the reset...
If I engrave a circle at S80, everything is ok... if I don't change the power all the engravings are successful. If I engrave another circle changing the power, the problem I described before is triggered! The problem remains until I reset the board!

terjeio, 2025-04-03:
> Continuing to do some tests, it seems that the enable output does not activate until inside the engraving it is given an S0 for the first time!!

gcode please, at least the initial lines up to the first expected laser on.

> If $34 is greater than zero, M5 or M2 DO NOT turn off the PWM at the end of the process....

Correct - same as the behaviour from the [LPC version of Grbl](https://github.com/gnea/grbl-LPC/blob/2254e9f85aaae96c553e776d7b13ffad8374cbc6/grbl/spindle_control.c#L143-L146) that has the $34 - 35 settings. 8-bit Grbl does not have them.

Deltaplc, 2025-04-03:
Resetting the board, the problem disappears, at the first engravings.... It seems that it is triggered between a single engraving and the other!
I try to explain:
If I engrave the first time after the reset, everything is ok... if I make other engravings, the problem I described before occurs! The problem remains until I reset the board!

Deltaplc, 2025-04-03:
> Corretto: lo stesso comportamento della [versione LPC di Grbl](https://github.com/gnea/grbl-LPC/blob/2254e9f85aaae96c553e776d7b13ffad8374cbc6/grbl/spindle_control.c#L143-L146) che ha le impostazioni $34 - 35. Grbl a 8 bit non le ha.

Yes of course, but in that version since there is no possibility that the pwm remains higher than 0 it could be right.... but $34-35 can make it so that at the end of the processing the pwm remains high, and in my opinion it is not safe... At least the M2 command (End of cycle) should turn the pwm off.
At the moment it is not essential, the machine works correctly anyway, but as I already said in my opinion turning the pwm off would be safer.

Deltaplc, 2025-04-03:
This is another test of cutting a simple square, with "cut through" enabled in ligthburn. Which creates a shot at the beginning of the cut.
In this case only the initial shot is performed, then the pwm remains high but the enable output does not 

G00 G17 G40 G21 G54
G90
; GCode clustering enabled (max 16)
G4 P1
M4
; Cut @ 900 mm/min, 76% power
M8
G0 X18.949Y176.01
M3
M3
G1 F100 S800
G4 P0.005
G1 S0
G0
M3
; Layer C00
G1 X4.949S760F900
G1 Y190.01
G1 X18.949
G1 Y176.01
G1 X18.449
M9
G1 S0
M5
G90
; return to user-defined finish pos
G0 X0Y0

Deltaplc, 2025-04-03:
> Il gcode che hai postato non incide in modalità M4, ma taglia in modalità M3. Devo verificare tutte le combinazioni di impostazioni e modalità prima di effettuare le modifiche.

Yes exactly, I always use the M3 mode, because I work better with my engraver!

> Quindi $34 dovrebbe essere rimosso? $31 imposta implicitamente il valore PWM minimo in ogni caso: avere due impostazioni che sostanzialmente fanno la stessa cosa è fonte di confusione: quale delle due dovrebbe essere utilizzata?

No no, don't delete either of them!
$34 and $31 are very different from each other! And they behave differently!
Why do you ask me this.....?

terjeio, 2025-04-03:
Try this:

[co2.zip](https://github.com/user-attachments/files/19587052/co2.zip)

If the Arduino does not mess up compilation you should get a CO2 plugin:

```
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:SAM3X8E]
[DRIVER VERSION:250327]
[BOARD:Ramps 1.6]
[AUX IO:4,3,0,0]
[PLUGIN:CO2 Laser v0.01]
```

When a laser spindle is enabled and RPM controlled enable is selected `M111P<n>` controls the PWM output during the off period. `<n>` is percent of the programmed S value. The `M111` value is reset to zero on program end (M2, M30) or on a soft reset. `M111` is just a random M-code I selected, I have no idea about what it should be in a final version...

Cutting should be ok with this version.

Deltaplc, 2025-04-03:
> When a laser spindle is enabled and RPM controlled enable is selected `M111P<n>` controls the PWM output during the off period. `<n>` is percent of the programmed S value. The `M111` value is reset [...]

Tell me if I understand correctly...if I am engraving at S80 and I set M111P10, the S value in the shutdown period becomes 80+10%?

> If the Arduino does not mess up compilation you should get a CO2 plugin:

I compiled it successfully, it seems to work on the bench, now I'll try to load it into the engraver and do some tests!

terjeio, 2025-04-03:
> Tell me if I understand correctly...if I am engraving at S80 and I set M111P10, the S value in the shutdown period becomes 80+10%?

Correct. How that affects the slope of delivered power depends on the response time (to PWM changes) of the laser PSU.

Deltaplc, 2025-04-03:
EVERYTHING SEEMS TO BE OK!!!
I attach the photo and 3 images engraved on white mdf.

1. Number one is engraved at 340mm/s 8.8power M111P58
2. Number two is engraved at 340mm/s 8.8power without M111 off
3. Number 3 is engraved at 340mm/s 10.8power without M111 off

As you can see the difference between the first two at the same power with and without M111 is abysmal!
In the third, by increasing the power you get much more details than on number two....
But compared to the first one the "black" parts are much darker and many details are lost, the line near the nose for example and many others!!
As you can see I didn't let you work without results!!!
I'm joking, of course!
In my opinion this whole strategy brings a great advantage to all those who have a Co2 laser and want to engrave images!!
I also did some "normal" engraving and cutting tests, it seems that everything behaves correctly!
In these days I will continue to do tests and if necessary I will update you on any problems!
Thank you so much for your work and support!!

[image]

[image]

Deltaplc, 2025-04-03:
[image]

Please add a donation button somewhere!!!
I would love to say thank you and buy you a beer or coffee!!!
Thanks again!!

terjeio, 2025-04-06:
Now [switched to M129](https://github.com/grblHAL/Plugins_laser#co2-laser-overdrive-for-dithered-engravings) to enable overdrive.

Deltaplc, 2025-04-06:
> Ora [sono passato a M129](https://github.com/grblHAL/Plugins_laser#co2-laser-overdrive-for-dithered-engravings) per abilitare l'overdrive.

Good morning
Ok, that's great!
Have you already inserted it in the plugins on all cors? Can it already be downloaded by everyone, even for manual compilation?

In these days I had the opportunity to do a lot of tests with the new plugin!
I have not found any negative aspects!!
No bugs!
It has improved in all aspects the behavior of my engraver! Photo engravings, written engravings and also on the cut!! No type of exit and entry ramp on full engravings!
Fantastic!
Again, many thanks for your work and your support!!

Deltaplc, 2025-04-06:
> Now [switched to M129](https://github.com/grblHAL/Plugins_laser#co2-laser-overdrive-for-dithered-engravings) to enable overdrive.

I downloaded the new version...
It seems to me there is a problem!! Am I doing something wrong?

[VER:1.1f.20250405:]
[OPT:NM0SL,100,1024,3,0]
Target buffer size found
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,EXPR,SED]
[FIRMWARE:grblHAL]
[SIGNALS:RHSP]
[NVS STORAGE:*FLASH]
[DRIVER:SAM3X8E]
[DRIVER VERSION:250403]
[BOARD:Ramps 1.4_Davide]
[AUX IO:3,3,0,0]
[CLUSTER:16]
Cluster size found
[PLUGIN:LightBurn clusters v0.06]
[PLUGIN:CO2 laser overdrive v0.01]
[PLUGIN:Feed override v0.01]
[PLUGIN:Events plugin v0.05]
ok
$9
$9=3
ok
$32
$32=1
ok
M129P30
error:20
Comando g-code non supportato o non valido trovato nel blocco.
M129 P30
error:20
Comando g-code non supportato o non valido trovato nel blocco.

closed 2025-04-06 by Deltaplc

Deltaplc, 2025-04-06:
> Now [switched to M129](https://github.com/grblHAL/Plugins_laser#co2-laser-overdrive-for-dithered-engravings) to enable overdrive.

I sent you a message, because it did not accept the command and gave me:

M129P30
error:20
Unsupported or invalid g-code command found in block.

Sorry but I had to reset the board several times, and remove power for it to accept the M129 command.....
Now it seems to work!!!

distebia, 2025-04-07:
I can't send m129p30, I generated from updated web build, where am I wrong?

Deltaplc, 2025-04-07:
Does it give you error 20?
Try resetting the board several times, and maybe even removing power. It did the same to me right after downloading the firmware...

distebia, 2025-04-07:
error 20...can't send m129 and can't find parameter $709, if I remember correctly. even resetting and restarting doesn't change anything

Deltaplc, 2025-04-07:
> error 20...can't send m129 and can't find parameter $709, if I remember correctly. even resetting and restarting doesn't change anything

Are you sure you have successfully compiled and uploaded the firmware? Make a $I and share the report!!

distebia, 2025-04-07:
[image]

terjeio, 2025-04-07:
> M129P30
error:20
Unsupported or invalid g-code command found in block.

I have added trapping of settings changes in todays commit so changing these ($39 and $9) will instantly make M129 available/unavailable.

The Web Builder has been updated to support the new plugin, check _CO2 laser overdrive_ in the _Plugin_ tab.

> but is the modification already available for all drivers?

For most, all supported by the Web Builder. There are a few (one?) boards that gates the PWM signal electronically - these will not work. Some has isolated PWM output via slow optocouplers, these may also be a bit problematic... And ensure the PWM frequency is high enough if doing high speed engraving.

$709 is only available when the driver and board supports two PWM spindles, and the second is enabled. Very few boards do.

distebia, 2025-04-07:
[image]

[image]

I used web build for the firmware with your latest modification, loaded the firmware, I see the plugin active but m129 is not accepted, both by changing $39 and $9=1, 3 and 7

[settings.txt](https://github.com/user-attachments/files/19637910/settings.txt)

Deltaplc, 2025-04-07:
In my case $32=1 $9=3 works fine!
For M129P(...) You have to reset the aboard after downloading, unplug the board (usb) and reconnect.....In my case, this procedure worked every time I compiled and downloaded the firmware...

distebia, 2025-04-07:
done + times, the plugin is present but I don't have the m129 commands...$32=1, $9=3, but $39 how should it be? mine is a normal co2 40w but with a teensy board, how do you keep the other parameters $32 etc.. for the laser?

Deltaplc, 2025-04-07:
$39=1
With this plugin the pwm always remains constant....the enable pin turns the laser on and off!!
I have no problem during overscan....
These are some tests...!
[Image](https://github.com/user-attachments/assets/fb8d9a44-be4f-42d1-9f96-66c9e8934ad2)

distebia, 2025-04-07:
congratulations for the photos...and how do you set the values ​​33, 34, 35 and 36? and the m129 command?

Deltaplc, 2025-04-08:
> values ​​33, 34, 35 and 36?

default settings

> and the m129 command?

Based on your engraver's response needs

distebia, 2025-04-08:
i miei valori sono $33=20000, $34=0, $35=7 e $36=100, il tuo 34 lo lasci a 0 e alzi solo da m129?

Deltaplc, 2025-04-08:
> il tuo 34 lo lasci a 0 e alzi solo da m129?

Yes, exactly.

> i miei valori sono $33=20000, $34=0, $35=7 e $36=100,

In my case $35=0
These are all values ​​that can vary from engraver to engraver! You will have to do some tests to see which configuration will work best for you!

distebia, 2025-04-08:
se faccio un cerchio con il valore $35=0 non viene chiuso, devo alzare a 7 per avere il cerchio chiuso...grazie per i consigli...

terjeio, 2025-04-09:
>  I notice that the laser does not turn off during overscanning and creates an external frame, as in the photo

Be aware that some boards has open collector/open drain outputs for PWM and/or spindle on. If not properly loaded these may exhibit rather low slew rates that can affect high speed engraving.

Deltaplc, 2025-04-10:
This is the essence of this plugin!!!!
Constant PWM, laser on/off through the Enable pin!!!
If you do not connect the laser power supply correctly, with both pins working, this plugin will not work!!!
The correct way to control the power supply is described by the manufacturer!!! You do not have to follow diagrams found online created by who knows who!!!

Pin IN= pwm
Pin L = Enable

[image]

@distebia  Tu sei italiano,giusto?

Deltaplc, 2025-04-10:
A tip....
Pay attention to the compatibility of the voltages between your board and the laser power supply!
In my case with an Arduino Due (3.3v), to be safe I made the coupling via an optocoupler made specifically for this job.

[8 more replies]