grblHAL/core #728: gSender Edge  - Hide A-axis controls when 4-axis or Rotary is not enabled

siganberg, 2025-04-14:
If 4-axis or Rotary is not enabled in the Rotary Tab or Settings, all A-axis controls should be hidden from the UI. Currently, these controls are still visible even though they have no function unless Rotary is enabled.

This can be confusing for users who don’t use a 4th axis or rotary setup, as it introduces unnecessary UI elements and raises questions about functionality that doesn't apply to their machine.

**Suggested Behavior:**

When Rotary mode is disabled in settings:

Hide all A-axis controls in the UI.

When Rotary mode is enabled:

Show A-axis controls as usual.

terjeio, 2025-04-15:
Why build with 4 axes enabled if you are not using it? Or do you want to hide A when a rotary is not connected? If so then that should be controlled by a different setting or a M-code. Or even detected by a signal.
The only reason I can think of is when using the fourth as a stepper spindle. And I have it on my todo list to handle that, along with user defined axis letters for axes 4+.

siganberg, 2025-04-15:
Opps I filed this issue on wrong repository. It supposed to be on the gSender.

closed 2025-04-15 by siganberg