grblHAL/core #731: Ethernet connection to SLB not working as expected

MattMead166, 2025-04-19:
Hello Terje, 

After Flashing my most recent FW build from the web builder I am having a strange issue with connecting to my SLB. I use an Ethernet connection from my Sienci Labs gControl computer and never had any issues. Since flashing when I go to connect through Ethernet it says "Connected" at the top but ALL my buttons remain greyed out and I can not control the CNC or open the FW tab. I just get a message in the console saying " connected to 192.168.5.1 at a baud rate of 115200" and that's it. anything I type in the console does nothing at all and the machine doesn't react. The only way I have found to get it "working" at this point is I leave the Ethernet Cable plugged in and the connection on Ethernet still, I just take a USB plug connected to my gControl computer and plug it in to the USB connection on the SLB and I instantly get the "connection sound/beep" and all my buttons ungrey and turn normal, and I can move the CNC and access the FW tab as normal. I don't do anything else and just "unplug" the USB cable from the SLB and it remains connected and working. I can disconnect and reconnect like normal with no issues until the SLB is rebooted/power cycled and then i have to go through all those steps again. Is there anything you can thing of that might be causing this issue? Steps to take? Settings to change? I have reached out to Chris at Sienci as well and he has no idea what the problem could be. I have changed to multiple different computers and cables and the problem persists on them all.

MattMead166, 2025-04-19:
I checked the IP as you mentioned and it was indeed the same, and checked $301 and $302 and they were already both set correctly no changes needed. I am stumped at this point. I did recall a windows update to my computer at the time of this issue occurring, but I have rolled that back with the same results. Additionally you can not flash the SLB with Ethernet, ONLY USB and I always unplugged my Ethernet cable, flashed through USB and then reconnected Ethernet. I noticed that I did NOT do this when the issue occurred. Apparently I was rushing and had a brain fart and flashed the SLB through USB but while simultaneously still having Ethernet plugged into the SLB and my computer. Do you think that somehow messed something up? My order of operations for today are as follows: I am going to flash the same FW I did when this issue occurred this time with ALL Ethernet unplugged and see if that resolves the issue. If not, then I am going to try a FW built last night with Virtual (UART) connection selected instead of Native USB as per a friends recommendation and see if that fixes the issue. If not, then I am going to flash the stock SienciHAL FW of 5.0.5b and see if that fixes the issue. If so then it will let me know something is wrong with my Web Builder versions. If NOT then I am going to completely reset the Windows on this PC and reinstall to see if Drivers or settings that I can not figure out somehow got corrupted or deleted. Please let me know if you have any other advice or steps that I should be taking. Thank you.

terjeio, 2025-04-19:
I can connect to my SLB with gSender, this with fresh Web Builds, so my guess is that something is amiss with your settings or configuration. My build is just a plain one with nothing else than networking enabled. You have not told me what yours is, output from `$i` in the console may help. And a settings backup may also be helpful.

If this is the first time you flashed a standard build then you should be aware that the Sienci version of grblHAL has several non-standard settings that _may_ cause issues. So it is not a good idea to load a settings backup from the Sienci version. And if you have then try with `$RST=*` to reset settings to default and check the connection again.

There should be no need to reinstall Windows, and reflashing with ethernet connected is perfectly ok - I do it all the time when working on grblHAL.

MattMead166, 2025-04-19:
ok so after going through some of my steps mentioned above, (not all) I have successfully flashed back to stock SLB FW of 5.0.5b and ALL my issues are resolved and ethernet is working as expected even after power cycle. So it must be something I selected in the web build that messed it up. I am just very happy that I got it working as expected again as I thought I messed something up that was unfixable. As for flashing your web builds I have done it over a 100 times in the past 3 months never with an issue. I am using the Rapid Change ATC and Macros as well as Siganberg's Auto Dust Boot so I have to use your custom FW as SienciHAL doesnt work properly with the SD card or the M6 Passthrough as Siganberg has addresses to you (and you have fixed). That is what had me confused this time. I follow your update log almost daily and anytime I see something that pertains to the SLB's STM32F4xx chip i usually create a new build and flash as I see you adding things that pertain to Sienci often. Because I still cant switch successfully between my default MODBUS spindle and my Sienci Laser so I always try new FW builds to see if things work yet. I also haven't tried a rotary job with your FW so I don't know if that works either but i do have the A axis show up in my FW so that is a good thing. As for my settings I selected on the Web Builder I believe these are the settings I built with when everything went wrong :

{"driver":"STM32F4xx","URL":"https://github.com/grblHAL/STM32F4xx","board":"BOARD_LONGBOARD32","my_plugin":["https://github.com/grblHAL/Templates/tree/master/my_plugin/Persistent_tool"],"symbols":["USE_HAL_DRIVER","ETH_TX_DESC_CNT=12","TCP_MSS=1460","TCP_SND_BUF=5840","LWIP_NUM_NETIF_CLIENT_DATA=2","LWIP_HTTPD_CUSTOM_FILES=0","MEM_SIZE=16384","LWIP_IGMP=1","LWIP_MDNS_RESPONDER=1","LWIP_NETIF_STATUS_CALLBACK=1","LWIP_HTTPD_DYNAMIC_HEADERS=1","LWIP_HTTPD_DYNAMIC_FILE_READ=1","LWIP_HTTPD_SUPPORT_V09=0","LWIP_HTTPD_SUPPORT_11_KEEPALIVE=1","LWIP_HTTPD_CGI_ADV=1","LWIP_HTTPD_SUPPORT_POST=1","LWIP_HTTPD_SUPPORT_WEBDAV=1","STM32F412Vx","STEP_PULSE_LATENCY=1.3","TRINAMIC_DYNAMIC_CURRENT=1","RTC_ENABLE=1","TMC_X_ENABLE=1","TMC_Y_ENABLE=1","TMC_Z_ENABLE=1","USB_SERIAL_CDC=1","N_AXIS=4","Y_GANGED=1","PROBE_ENABLE=1","TRINAMIC_ENABLE=2660","TRINAMIC_R_SENSE=50","TRINAMIC_SPI_ENABLE=2","MODBUS_ENABLE=1","MODBUS_BAUDRATE=3","EEPROM_ENABLE=128","SDCARD_ENABLE=1","MACROS_ENABLE=1","N_MACROS=3","RGB_LED_ENABLE=2","EVENTOUT_ENABLE=1","N_EVENTS=4","PPI_ENABLE=1","LB_CLUSTERS_ENABLE=1","HOMING_PULLOFF_ENABLE=1","FEED_OVERRIDE_ENABLE=1","ODOMETER_ENABLE=1","_WIZCHIP_=5500","ETHERNET_ENABLE=1","TELNET_ENABLE=1","WEBSOCKET_ENABLE=1","FTP_ENABLE=1","ESTOP_ENABLE=1","N_TOOLS=8","MINIMIZE_PROBE_OVERSHOOT=1","NGC_EXPRESSIONS_ENABLE=1","ROTARY_FIX=1","PROBE_PROTECT_ENABLE=1","STATUS_LIGHT_ENABLE=2","ADD_MY_PLUGIN=1","LIMITS_OVERRIDE_ENABLE=1","MCP3221_ENABLE=1","SPINDLE0_ENABLE=11","SPINDLE1_ENABLE=5","SPINDLE2_ENABLE=14","N_SPINDLE=3"],"docker_instance":"","grblHAL":"webBuilder"}

MattMead166, 2025-04-19:
now getting this when trying to build today. 

Command failed: pio run -etest -cjob.ini
motors/trinamic.c:562:17: warning: 'get_driver_enable' defined but not used [-Wunused-function]
static uint32_t get_driver_enable (setting_id_t setting)
^~~~~~~~~~~~~~~~~
motors/trinamic.c:549:22: warning: 'set_driver_enable' defined but not used [-Wunused-function]
static status_code_t set_driver_enable (setting_id_t id, uint_fast16_t value)
^~~~~~~~~~~~~~~~~
networking/wiznet/port/w5x00_lwip.c:167:17: warning: 'ethernet_frame_crc' defined but not used [-Wunused-function]
static uint32_t ethernet_frame_crc(const uint8_t *data, int length)
^~~~~~~~~~~~~~~~~~
In file included from networking/wiznet/enet.c:55:0:
networking/wiznet/port/w5x00_lwip.h:110:17: warning: 'ethernet_frame_crc' declared 'static' but never defined [-Wunused-function]
static uint32_t ethernet_frame_crc(const uint8_t *data, int length);
^~~~~~~~~~~~~~~~~~
.pio/libdeps/test/grblhal-rgb-plugin/rgb.c: In function 'RGBUpdateState':
.pio/libdeps/test/grblhal-rgb-plugin/rgb.c:213:40: warning: passing argument 2 of 'task_add_delayed' makes pointer from integer without a cast [-Wint-conversion]
task_add_delayed(set_hold, 200, NULL);
^~~
In file included from ./grbl/protocol.h:27:0,
from .pio/libdeps/test/grblhal-rgb-plugin/rgb.c:43:
./grbl/task.h:28:6: note: expected 'void *' but argument is of type 'int'
bool task_add_delayed (foreground_task_ptr fn, void *data, uint32_t delay_ms);
^~~~~~~~~~~~~~~~
.pio/libdeps/test/grblhal-rgb-plugin/rgb.c:213:45: warning: passing argument 3 of 'task_add_delayed' makes integer from pointer without a cast [-Wint-conversion]
task_add_delayed(set_hold, 200, NULL);
^~~~
In file included from ./grbl/protocol.h:27:0,
from .pio/libdeps/test/grblhal-rgb-plugin/rgb.c:43:
./grbl/task.h:28:6: note: expected 'uint32_t {aka long unsigned int}' but argument is of type 'void *'
bool task_add_delayed (foreground_task_ptr fn, void *data, uint32_t delay_ms);
^~~~~~~~~~~~~~~~
.pio/libdeps/test/grblhal-rgb-plugin/rgb.c: In function 'status_light_init':
.pio/libdeps/test/grblhal-rgb-plugin/rgb.c:363:9: warning: implicit declaration of function 'protocol_enqueue_rt_command'; did you mean 'protocol_enqueue_realtime_command'? [-Wimplicit-function-declaration]
protocol_enqueue_rt_command(RGBUpdateState);
^~~~~~~~~~~~~~~~~~~~~~~~~~~
protocol_enqueue_realtime_command
.pio/build/test/lib81a/grblhal-rgb-plugin/rgb.o: In function `status_light_init':
rgb.c:(.text.status_light_init+0x74): undefined reference to `protocol_enqueue_rt_command'
collect2: error: ld returned 1 exit status
*** [.pio/build/test/firmware.elf] Error 1
========================== [FAILED] Took 4.67 seconds ==========================

MattMead166, 2025-04-19:
It seems I can select ALL my normal options i always select but once I add "Sienci Lights" is when it fails. whether or not I have LED strip config + 150 selected or not. In ALL my previous builds I have always selected LED strip config + 150 and then turned on "Sienci Lights" and never had an issue. Now it will not let me build if "Sienci Lights" is selected. I went through every option one by one and built each time and was successful until "Sienci Lights" then I cant build and get all  kinds of RGB errors. 

Command failed: pio run -etest -cjob.ini
motors/trinamic.c:562:17: warning: 'get_driver_enable' defined but not used [-Wunused-function]
static uint32_t get_driver_enable (setting_id_t setting)
^~~~~~~~~~~~~~~~~
motors/trinamic.c:549:22: warning: 'set_driver_enable' defined but not used [-Wunused-function]
static status_code_t set_driver_enable (setting_id_t id, uint_fast16_t value)
^~~~~~~~~~~~~~~~~
plugins/eventout.c: In function 'event_settings_restore':
plugins/eventout.c:308:38: warning: array subscript is above array bounds [-Warray-bounds]
plugin_settings.event[idx].trigger = Event_Ignore;
~~~~~~~~~~~~~~~~~~~~~^~~~~
In file included from networking/wiznet/enet.c:55:0:
networking/wiznet/port/w5x00_lwip.h:110:17: warning: 'ethernet_frame_crc' declared 'static' but never defined [-Wunused-function]
static uint32_t ethernet_frame_crc(const uint8_t *data, int length);
^~~~~~~~~~~~~~~~~~
networking/wiznet/port/w5x00_lwip.c:167:17: warning: 'ethernet_frame_crc' defined but not used [-Wunused-function]
static uint32_t ethernet_frame_crc(const uint8_t *data, int length)
^~~~~~~~~~~~~~~~~~
.pio/libdeps/test/grblhal-rgb-plugin/rgb.c: In function 'RGBUpdateState':
.pio/libdeps/test/grblhal-rgb-plugin/rgb.c:213:40: warning: passing argument 2 of 'task_add_delayed' makes pointer from integer without a cast [-Wint-conversion]
task_add_delayed(set_hold, 200, NULL);
^~~
In file included from ./grbl/protocol.h:27:0,
from .pio/libdeps/test/grblhal-rgb-plugin/rgb.c:43:
./grbl/task.h:28:6: note: expected 'void *' but argument is of type 'int'
bool task_add_delayed (foreground_task_ptr fn, void *data, uint32_t delay_ms);
^~~~~~~~~~~~~~~~
.pio/libdeps/test/grblhal-rgb-plugin/rgb.c:213:45: warning: passing argument 3 of 'task_add_delayed' makes integer from pointer without a cast [-Wint-conversion]
task_add_delayed(set_hold, 200, NULL);
^~~~
In file included from ./grbl/protocol.h:27:0,
from .pio/libdeps/test/grblhal-rgb-plugin/rgb.c:43:
./grbl/task.h:28:6: note: expected 'uint32_t {aka long unsigned int}' but argument is of type 'void *'
bool task_add_delayed (foreground_task_ptr fn, void *data, uint32_t delay_ms);
^~~~~~~~~~~~~~~~
.pio/libdeps/test/grblhal-rgb-plugin/rgb.c: In function 'status_light_init':
.pio/libdeps/test/grblhal-rgb-plugin/rgb.c:363:9: warning: implicit declaration of function 'protocol_enqueue_rt_command'; did you mean 'protocol_enqueue_realtime_command'? [-Wimplicit-function-declaration]
protocol_enqueue_rt_command(RGBUpdateState);
^~~~~~~~~~~~~~~~~~~~~~~~~~~
protocol_enqueue_realtime_command
.pio/build/test/lib81a/grblhal-rgb-plugin/rgb.o: In function `status_light_init':
rgb.c:(.text.status_light_init+0x74): undefined reference to `protocol_enqueue_rt_command'
collect2: error: ld returned 1 exit status
*** [.pio/build/test/firmware.elf] Error 1
========================== [FAILED] Took 4.68 seconds ==========================

MattMead166, 2025-04-20:
So after multiple FW tests its definitely "Sienci Lights" that are causing the errors. If I leave it disabled or Datron style it works fine.

MattMead166, 2025-04-20:
No matter what FW options I select to get a successful build though they ALL result in the Ethernet NOT working without plugging in USB first. If I am on the Stock SienciHAL FW of 5.0.5b I have NO issues at all except for the fact that I can now NOT use my ATC with Macros as it gives an error when it triggers the IR sensor.

terjeio, 2025-04-20:
> I am using the Rapid Change ATC and Macros as well as Siganberg's Auto Dust Boot so I have to use your custom FW as SienciHAL doesnt work properly with the SD card or the M6 Passthrough as Siganberg has addresses to you (and you have fixed).

Why not get Sienci to fix their official version then? I am not obliged to create custom versions that works, and for free...

>  Because I still cant switch successfully between my default MODBUS spindle and my Sienci Laser so I always try new FW builds to see if things work yet.

I do not have the capacity, nor equipment, to test every possible combination of options. So I rely on feedback, if I do not get any you cannot expect issues that I have never encountered to be fixed.

FYI I have now replaced some function calls, that has been flagged as deprecated for years, in 3rd party code to make it compile again. With this in place I can connect via ethernet without the USB cable plugged in, both with ioSender and gSender. And I used your build options (I loaded the json above).

BTW you should get yourself a new keyboard with a working Enter key - your comments are somwehat hard to read.

MattMead166, 2025-04-20:
Apologies if my messages came in weird? Everything looks fine and easily readable on my side. I very much appreciate you and ALL the work that you do for the community. I know you must have people messaging you all day everyday. 

I have asked Chris and the team at Sienci multiple times to fix these issues and unfortunately have been told they do not have the capacity at this time due to other ongoing projects. I have Siganbergs FW to use also but he doesn't use a Laser or Rotary so the options arent even in that build.  

I absolutely wouldn't expect you to or even be able to test all the combinations of setups out there and very much appreciate the work you do for the community. Apologies if it came off any other way. I was pretty sure I mentioned about the Spindle / Laser switching errors in my last issue I had posted here, and Chris at Sienci said he was going to reach out about it also. If I didn't mention and he didn't reach out I apologize. 

Yes I just checked the web builder again and was indeed able to compile from my previous JSON file correctly with "Sienci Lights" enabled. I will flash it in the morning and see if the issue persists. Thank you for making those changes. Did you see anything that was wrong in my JSON that I should NOT be putting in my build?

terjeio, 2025-04-22:
>  I can not connect to the SLB through Ethernet any longer without connecting a USB cable first to finish the Handshake. 

Only connect the cable or do you have to connect a client as well? At my end I can connect via ethernet wit the USB cable out so what might be different from your setup should only be the settings. So please export them, zip and post in a comment.

> I did notice when I did $I from the stock FW and $I from the grblHAL FW that the section for [NETCON: Telnet] is MISSING from the grblHAL FW, even after confirming Static IP is selected and power cycling the SLB.

`[NETCON: Telnet]` will only show up if there is a client connected via ethernet/telnet.

MattMead166, 2025-04-22:
So I was connected via Ethernet that is why I expected to see [NETCON: Telnet] in the console?? I have also noticed that it takes multiple attempts of flashing, $rst=$ and power cycling to get the FW fully flashed and loaded as expected. I think that might be what is going on?

I loaded a copy of Siganberg's FW yesterday and it worked perfectly with Ethernet but did NOT have my correct VFD selected. I wrote new FW using his JSON as a base and selected the correct spindles, created and flashed. Same issues as before with the Ethernet/USB. 

I then reflashed Siganberg's original FW once more and this time it didn't work also and Ethernet/USB issue was present again even though it worked perfectly the first time. 

So I think i just need to keep forcing the flash and $rst=$ until it fully resets and flashed correctly. Cause I did prove that it worked as expected the first time I flashed this morning. Siganberg mentioned to me last night he also has had issues flashing grblHAL with gSender and needed to do it multiple times so i will give that a shot today.

MattMead166, 2025-04-22:
[SETTINGS - MATT MEAD_WORKING_grblHAL FW for RC ATC_with ROTARY_LASER NOT WORKING PROPERLY_CREATED 20250129_gSender-settings-3_6_2025-4_06_42 PM.zip](https://github.com/user-attachments/files/19853978/SETTINGS.-.MATT.MEAD_WORKING_grblHAL.FW.for.RC.ATC_with.ROTARY_LASER.NOT.WORKING.PROPERLY_CREATED.20250129_gSender-settings-3_6_2025-4_06_42.PM.zip)
[FW - MATT MEAD_WORKING_grblHAL FW for RC ATC_with ROTARY_LASER NOT WORKING PROPERLY_CREATED 20250129_gSender-firmware-settings-3_6_2025-4_00_07 PM.zip](https://github.com/user-attachments/files/19853977/FW.-.MATT.MEAD_WORKING_grblHAL.FW.for.RC.ATC_with.ROTARY_LASER.NOT.WORKING.PROPERLY_CREATED.20250129_gSender-firmware-settings-3_6_2025-4_00_07.PM.zip)

terjeio, 2025-04-22:
It seems gSender does not work properly if more than one spindle is enabled, why I do not know. I added a new capability letter to the `$spindlesh` output 2024-03-09 which should not matter if parsed correctly - again I do not know if this is the culprit or if the output is used at all. From the console it looks like the human readable version, `$spindles`, is used.

Anyway I have no problem with _connecting_ when the USB cable is out, even when flashing via gSender.

Switching between the laser and VFD spindle looks like a gSender issue, it sends $-settings to switch bewteen them - which must be dependent on custom Sienci code. The grblHAL way is to use `M104` to [switch between spindles](https://github.com/grblHAL/Plugins_spindle/#spindle-selection), and it is not always neccesary to change $-settings. E.g. `$32=1` (laser mode) can be left permanently on and is automatically switched based on spindle laser capability. The RPM range is read from the VFD if the VFD driver supports it so the RPM settings can often be left as-is too.

If you are running gSender on Windows, or have access to a Windows machine, you can try [ioSender](https://github.com/terjeio/ioSender) to check if the controller is misbehaving or gSender is. If ioSender works then I suspect your problems are related to gSender.

BTW from your settings and Web Builder file I see that you have configured two PWM spindles, do you have three spindles on your machine?

MattMead166, 2025-04-22:
Thank you for the updates Terje, I really appreciate all you help. I do have a Windows machine and will test on ioSender as mentioned. I definitely see the error of more then one spindle selected cause if I leave default spindle to MODVFD and the others disabled then all seems to be ok and I have X,Y,Z and A axis $ settings in FW as expected. But if I have default as MODVFD and select PWM2 for spindle 2 that when the FW goes nuts, and after power cycling then all my A axis $ settings are gone from FW. 

I believe the Laser and VFD spindle is indeed a gSender issue and have addressed this to Chris and the team. I am hoping they can get this figured out and resolved as soon as possible. That may be part of the issue also as if I try to change the tool number for spindle 2 then it also changes spindle 1 also to the same number. When selecting the Laser tab and hitting the drop down ALL spindles are listed there but ALL have the same tool number and are highlighted blue together and will not let me select. 

As for the spindles I only have one spindle and one laser. It was my understanding that PWM needed to be the first spindle but i may be incorrect. I have a MODVFD spindle and was told by Chris that the laser function is the PWM2 (no direction) Is that not correct either?

MattMead166, 2025-04-23:
Ok so I downloaded and installed ioSender as mentioned. Unfortunately it has the same issue and would NOT connect to my machine through Ethernet. 

It gave an error stating "No response received from controller, exiting" and then closed ioSender. Once i plugged in my USB and heard the beep I opened ioSender again and selected "networking" and confirmed my IP and connected without issue. Same as gSender I could then disconnect the USB and remain on Ethernet. 

This remains until I power cycle the SLB and then I must do it all over again. This is the FW I built and currently have loaded attached below. 

[STM32F4xx_LONGBOARD32_FRANCIS ONE FINITY_MODVFD-PWM2_3 MACROS TO AUX_8 TOOL TABLE_4 EVENTS_LASER PPI - LIGHTBURN_FEED RATE OVERRIDE_SIENCI LIGHTS_KEEP LAST TOOL_.json](https://github.com/user-attachments/files/19859478/STM32F4xx_LONGBOARD32_FRANCIS.ONE.FINITY_MODVFD-PWM2_3.MACROS.TO.AUX_8.TOOL.TABLE_4.EVENTS_LASER.PPI.-.LIGHTBURN_FEED.RATE.OVERRIDE_SIENCI.LIGHTS_KEEP.LAST.TOOL_.json)

terjeio, 2025-04-23:
FYI I have a SLB which I believe is a late prototope, not a production version. Perhaps there is some differences in how stuff is powered? Since building with the .json above still works for me.

Can you try powering the SLB via USB using a telephone charger? Does that allow you to connect?

> As for the spindles I only have one spindle and one laser. It was my understanding that PWM needed to be the first spindle but i may be incorrect.

You can select any spindle you want from the _Spindle 1_ dropdown. There is only one restriction, if using the standard PWM spindle it has to be selected as _Spindle 1_,

> I have a MODVFD spindle and was told by Chris that the laser function is the PWM2 (no direction) Is that not correct either?

That is correct.

> That may be part of the issue also as if I try to change the tool number for spindle 2 then it also changes spindle 1 also to the same number. 

There is only one tool number slot shared between the spindles so this is normal.

>  When selecting the Laser tab and hitting the drop down ALL spindles are listed there but ALL have the same tool number and are highlighted blue together and will not let me select.

Hmm, only the MODVFD spindles shows up for me with _Default_ as the value in the drop-down. And I cannot select either...
Spindle selection in ioSender works.

> But if I have default as MODVFD and select PWM2 for spindle 2 that when the FW goes nuts, and after power cycling then all my A axis $ settings are gone from FW.

Does not happen for me when using ioSender. Either gSender messes up or you have a defective SLB?

MattMead166, 2025-04-24:
Hello again Terje, 

So I tried as you mentioned to connect to the SLB with a "phone charger" that was plugged into a 5v power block to see if that allowed the Ethernet connection and it did NOT work. ONLY a USB from computer to SLB completes the wake up handshake and allows the Ethernet to work. 

I never seemed to have an issue prior until the grblHAL DRIVER VERSION went to 250412. I have a new computer now as my old one crashed and I'm unable to recover the files. My old PC though had a grblHAL FW version that the DRIVER VERSION was something like 250319 (or close to that) that worked perfectly (minus the switching between Spindle/Laser issue). I also have one from the first batch of 500 SLB's.

I know my SLB is NOT defective though because I can flash the OEM Sienci Labs FW of 5.0.5b and 5.0.5b_NGC and both FW's work perfectly for Ethernet connection without the need for a USB cable first to complete the handshake. As mentioned though they do NOT work fully for my RCATC setup as they do not have your new paramaters of probe_state and toolsetter_state implemented, hence why I have been attempting to build a working grblHAL version. 

As you mentioned I do see the correct spindle and laser in ioSender but the Ethernet issue is still present with that software also so that doesnt really help my issue unfortunately. But is nice to be able to use the Laser with current FW and NOT have to flash back to Sienci OEM FW. 

At this point I am just confused why all my JSON builds work ok for your SLB and Ethernet connection but not mine? I thought it might have been some windows updates that may have affected my "Realtek PCIe gbe family controller" but as mentioned it works absolutely fine with the OEM Sienci FW so I do not belive that is the issue either? 

I just want to thank you though for the time you have spent investigating this and helping me to try and resolve my issue. If I could somehow get that old FW from the crashed computer I believe I would likely be up and running again as expected but my brother was unable to extract any data from it.

terjeio, 2025-04-24:
Can you build and test a version with just networking and telnet enabled?

MattMead166, 2025-04-25:
Do you mean like this in the attached picture? what about all the other options in the different tabs? does that matter? 

[image]

terjeio, 2025-04-25:
> Do you mean like this in the attached picture? what about all the other options in the different tabs? does that matter?

Yes, and no other options since it might be one of them that is causing your issue. If it is we have to find out which because I am not able to replicate.

MattMead166, 2025-04-25:
Ok understood, I can try that. Should I at least pick a Spindle though for Spindle 1 or leave it at the default PWM and just confirm that Ethernet connects to the SLB as expected?

terjeio, 2025-04-25:
Leave at default, if that works then add options.

MattMead166, 2025-04-25:
ok so with this JSON loaded below with ONLY PWM as default spindle and NETWORKING / TELNET activated I can NOT connect to Ethernet on the SLB at all now. Whether or not the USB is plugged in. Ethernet just spins and never connects. 

[STM32F4xx_LONGBOARD32 TERJE DEFAULT TEST.json](https://github.com/user-attachments/files/19917052/STM32F4xx_LONGBOARD32.TERJE.DEFAULT.TEST.json)

terjeio, 2025-04-25:
Please try this - from the Web Builder with your .json above, created _after deleting the build folder_:

[firmware-19.zip](https://github.com/user-attachments/files/19917597/firmware-19.zip)

It works for me, the one I created before deleting did not if the USB cable was unplugged. If it works for you then try a _new_ build with the builder.

> "likely something with the port initialization in the Firmware" 

It seems the Web Builder (PlatformIO based) can mess up the code generation, and that explains why I am not able to replicate when I build with the STM32CubeIDE. But not why my Web Builder builds worked before... So IMO there is a nasty Gremlin interfering and likely the code is ok?

MattMead166, 2025-04-26:
Unfortunately that FW did NOT work. Same issue with my default test FW. I was NOT able to connect to Ethernet at all, even with USB plugged in. Apologies but what do you mean when you say "after deleting the build folder"? maybe that is my issue? should i be deleting something first? 

I also am NOT able to ping 192.168.5.1 through TELNET in the cmd prompt with either of those FW's. As soon as I go back to SienciHAL FW I can connect through Ethernet as expected and also ping 192.168.5.1 through TELNET in the cmd promt. I was able to see the connection and also type $$ to show all my FW settings. 

I absolutely agree it must be some gremlin or something that is also causing issues, especially when builds work fine for you but NOT for me. Do you have a preference of flashing using gSender or the STMCube32 Programmer? 

I don't know if this is possible or even an option but would you be willing to charge me and have me pay you for your services to take the existing 5.0.5b SLB FW and fork it with the grblHAL FW to get the SD card, NGC expression flow and the probe_state and toolsetter_state parameters into a CUSTOM BUILT FW?? I know this is not an easy job (though it prob is for you) and I would NOT expect it for free and would absolutely pay at this point. If i could get a forked version of the SienciHAL with those options added i believe that would solve ALL of my issues. I would be able to use my MODVFD spindle, switch between my SLB laser and Rotary as well as be able to use my RCATC and all the macros without any issues or having to jump between FW's. I understand you are a very busy man and if this is NOT an option I totally understand but you never know unless you ask and at this point with all my issues I am totally willing to pay for someone's services to be able to finally use my machine again with all my upgrades and start making projects again.

terjeio, 2025-04-26:
> Do you have a preference of flashing using gSender or the STMCube32 Programmer?

I use the STM tools for flashing, both I have are working.

>  Apologies but what do you mean when you say "after deleting the build folder"? maybe that is my issue?

Could be - but not with the firmware I posted above. The build folder is on the Web Builder server.

Try the firmware I posted again. Connect via USB, reset the settings with `$RST=*` and change $301 to static IP. Unplug the USB cable and power cycle the controller, then try connect with [PuTTY](https://www.putty.org/):

[image]

Is connect successful? And do you get the welcome message like this:

[image]

If you get it send `$i` - any output?

And to be 100% sure - you unpacked the .zip before flashing?

---

An option would be to disable USB after configuring the controller, you can do this by selecting _Virtual (UART)_ for _Connection_:

[image]

and reflash. I do not think it would affect VFD communication, I have not tested.

However, I would rather root out the Gremlin...

MattMead166, 2025-04-26:
Hello Terje, 

Thank you for the new steps to test. Yes sir I indeed unpacked the .zip file before loading. I flashed your above FW again with gSender and with STMCube Programmer to be sure. I then tried to connect as instructed to Putty and was unable to connect as you can see in the picture below. Same thing when I try and connect through gSender or use the TELNET ping option in the cmd prompt. 

[image]

I will try and build a FW with UART as the connection type and see what happens. I already tried that prior and it didn't work as well but it was a fully loaded build and not just the defaults like we tried previously. 

Just wondering if you saw my last question in my previous reply? Would that be an option or you do not offer that service, or have the bandwidth to complete? Like I said you never know unless you ask. Either way though is totally understandable as I know you are very busy I'm sure.

terjeio, 2025-04-27:
> Just wondering if you saw my last question in my previous reply? Would that be an option or you do not offer that service, or have the bandwidth to complete?

That would take a lot of time and be a dead end.

I have spent more time on this and finally tested with a second ethernet port and a cable directly to the controller. With that I was able to replicate and from what I have found this far is that somehow the ethernet stack on the PC side can start to misbehave, this with my Win10 based development machine. Flashing with the ethernet cable connected _might_ be the reason after all. After shutting down the machine and starting up again (not rebooting) it started working again.
When it fails either the link is not established (cable unplugged) or I cannot connect due to network services not starting in the controller (they are started on link established when using a static address so somehow that event is not generated).
A bit odd is that I have not managed to replicate with my Win7 laptop, I can move the cable over and it immediately recognizes the controller and I can connect. Moving it back to to the Win10 machine and it does not work if id did not previously. Another strange thing is that the Win10 machine may take a long time to establish the link - several minutes in the worst case.

MattMead166, 2025-04-28:
Hello Terje, 

So finally a little bit of GOOD news to report for once. I just built a Test FW with the Web Builder (JSON attached below) and Ethernet connection works fully as expected. Even after power cycling the SLB multiple times and loading FW settings it is still working as expected. So that is definitely a positive finally. 

This has really NO options selected and pretty much ALL defaults with only MODVFD as the selected spindle (all else disabled) but it worked perfectly. Now that I have some positive results finally I am going to build my FW's only ONE option at a time, flashing in between each build until it all works as expected or I find the option that is making it go crazy so I can report back here with that info. Fingers crossed all goes well and there isn't any issues to report and the "Gremlin" is finally gone. 

[FULLY WORKING w_ETHERNET - STM32F4xx_LONGBOARD32 TEST WITH MODVFD_NO OTHER OPTIONS.json](https://github.com/user-attachments/files/19932603/FULLY.WORKING.w_ETHERNET.-.STM32F4xx_LONGBOARD32.TEST.WITH.MODVFD_NO.OTHER.OPTIONS.json)

MattMead166, 2025-04-28:
Well unfortunately the good news from yesterday was short lived. I power cycled multiple times and all was good yesterday. Turned my machine off and went to bed. But I came out to the shop just now and powered everything up and the issue is now back and I have not changed anything from the working version that was tested last night.

terjeio, 2025-04-29:
What is the network adapter status? _Network cable unplugged_ or _Unidentified network_?

_Control Panel > Network and Sharing Centre > Change adapter settings_

I had to shut down/restart my Win10  machine again to get it working, Win7 machine was up in seconds, no restart.

Do you have a switch (or router) at hand that you can put in between the controller and the computer? I think could be worth a try because I do not see the issue when I connect via a switch.

terjeio, 2025-04-30:
Some progress: Windows version and/or ethernet adaptor settings seemingly affects how the _link_ is established.

* Win7 laptop always establishes link (with the controller) and allows connect in a few seconds.
* Win10 machine connected via a switch likewise.
* Win10 machine with default adaptor settings randomly establishes link, reboot sometimes is required. 
* Win10 machine with modified adaptor settings seems to work as it should, link is established quickly and I have yet to se a failure:

[image]

[image]

[image]

After spending many hours on this issue I am fairly certain that USB _at the controller side_ has nothing to do with it and the reason it fails is that when the PC and the controller is _directly_ connected (not via a switch) both are waiting for the link to established. Why the link can sometimes be established and sometimes not I do not know. I suspect the PC side is more likely to be the culprit than the controller since the controller is a lot simpler system and should end up in the exact same state on bootup, at least when nothing is connected (no USB or ethernet cable).

MattMead166, 2025-05-03:
Hello Terje, 

My apologies for the delayed response, I was out of town for work. I saw that you have sent two responses that i have not answered since. 

From your first response: 
The network status is "unidentified network".

I always shut down and fully power down my computer and SLB when I am done for the day, I never leave it running or in sleep mode. Also I don't do soft resets usually either. 

Sorry, unfortunately I do not have an old router or switch to hook up in between the computer and SLB. My question though would be why would that be necessary to add extra devices when I can use the stock FW without issues? That seems counter productive too me and just a band-aid it would seem, no?

From your second response:
I tried to change those settings you mentioned above. Unfortunately I have Windows11 not 10 and I also have a Realtek PCIe Gbe Family Controller #2 Ethernet adapter and not an Intel like you do. I looked through all my settings and the only one I had was the "speed and duplex" setting and I changed that to match yours. Unfortunately the issues persist. I did not have a setting for "wait for link" just "wake on magic packet" and "wake on pattern match. Even after I updated the driver only new power saving options showed up. 

MY NEXT QUSTION:
I am attempting to compile the firmware manually instead of using the web builder to see if that fixes any issues possibly. I was able to follow the instructions on Sienci's GitHub Repo. I forked the STMF4xx repo recursively and it was successful. I loaded into VS code with the Platform.io extension. I was able to build all the FW's successfully and I see them all in the .pio folder but I do not see a version for the SLB? In there example they show below:

[image]

and there picture shows them using the nucleo_f411re_protoneer build which does build successfully and is in my list but if my assumptions are correct that is not right. That shows it using the board_build.ldscript = STM32F411CEUX_FLASH.ld, but from what I see the SLB and SLB-EXT should use the board_build.ldscript = STM32F412VGTX_FLASH.ld correct? If that is so I do not see a [env] present in the platformio.ini that is compatible with the SLB as they all have different versions. 

I then tried to "add" the SLB [env] to the platformio.ini file but I must of still done something wrong because it DID show up under my PROJECT TASKS as a board to build but it FAILED, while all others built and were a SUCCESS. Can you help me with what i did wrong here or might be missing so I can attempt a manual build and test that?

terjeio, 2025-05-04:
> The network status is "unidentified network".

Ok, that means the link is up. I have committed some further changes to link handling and also enhanced the `$netif` command to show listening ports, e.g:

`[MSG:IF=e00 IP=192.168.5.1 MAC=02:08:dc:07:8f:43 Listening=23]`

Port 23 is default for telnet - ports are only set listening after link is up and protocols are enabled. These changes are now available in the Web Builder. If port 23 is listening then you should be able to connect provided there is no firewall blocking the port.

>  My question though would be why would that be necessary to add extra devices when I can use the stock FW without issues? That seems counter productive too me and just a band-aid it would seem, no?

I have no good answer for this. The two PCs I have used for testing clearly behave differently with the same controller firmware, why I do not know - I do not have perfect knowledge of every detail of what is going on at every level in the network stack at both ends of the cable. And is frustrating to work with the Win10 machine since it works for a while and then the next morning it might not...

> Can you help me with what i did wrong here or might be missing so I can attempt a manual build and test that?

FYI I do not support platformio builds myself, I rely on pull requests to keep published platformio.ini up to date. However, the Web Builder uses platformio to build for most processors, a platformio.ini (job.ini) file is created based on the selections made. This ,ini-file can be downloaded with the build if you open the Web Builder with [this link](https://svn.io-engineering.com:8443/?dev=1). If no 3rd party plugins are enabled you should be able to use that without modifications.

MattMead166, 2025-05-05:
Hello Terje, 

I hope you had a good weekend? Thank you for the updated changes to the web builder, I will give that a shot today and see what happens and report back. Additionally, thank you for that other link that provides a platformio.ini file as well. I will also try that out. 

Since you don't support Platformio builds yourself, I assume you use STM32 Cube IDE correct? If so, might you be so kind as to guide me on how to build a SLB specific build using this IDE? I have downloaded and installed it but I have never used it before and don't want to mess anything up more then it currently is LOL. I would like to give a "manual build" a try as well as the web builder to see if the same issues arise or not.

terjeio, 2025-05-06:
> I hope you had a good weekend?

I was busy in the garden, and even spent a bit of time in the workshop! So no too bad.

> I assume you use STM32 Cube IDE correct?

Correct.

> If so, might you be so kind as to guide me on how to build a SLB specific build using this IDE?

[Import the project](https://github.com/grblHAL/core/wiki/Compiling-grblHAL#eclipse-based-ide) and edit _Inc/my_machine.h_ and _grbl/config.h_ to suit your needs. Select _Release F412_ (or _Debug F412_) from the "hammer" tool in the toolbar to compile - a new folder with the firmware should then appear if compilation succeds. For some unknown reason I sometimes have to compile twice before it appears.

MattMead166, 2025-05-14:
Hello Terje, 

I hope you are doing well sir? So I was able to figure out the platformio option and create a successful build. Unfortunately it yielded the same results as the web builder. The Ethernet connection originally connected right away then after a power cycle of the SLB, it went back to the same issues. I spoke with Chris at Sienci also and he said he was going to look into sending you a newer SLB to test against. Don't know if he has or not yet or if that will help or not?

Going a different route for testing, since the stock SienciHAL FW works fine for everything but doesn't have the new parameters you added for "call level" "probe_state" or "toolsetter_state". Could you help to direct me as to what specific files in grblHAL/core i would need to copy into SienciHAL's grbl secion to get these parameters into the code and working? I just want to make sure I make the correct edits to ALL files as I am sure it is more then just ngc_params.c/h. If you could give me some direction on which files contain the needed info so I can start to make edits on my end it would be much appreciated.

terjeio, 2025-05-15:
> I spoke with Chris at Sienci also and he said he was going to look into sending you a newer SLB to test against. Don't know if he has or not yet or if that will help or not?

I would be a lot more helpful if Sience (and/or other users) would try a Web Builder build since I suspect the network chip in the PC has something to do with this. Ideally breakpoints should be set for debugging to check which network events occur/do not occur.

>  If you could give me some direction on which files contain the needed info so I can start to make edits on my end it would be much appreciated.

It is basically ngc_params.c/h - but there are dependent changes elsewhere, at least gcode.c will have to be modified- not replaced. And likely ngc_expr.c and some other files as well. IMO reverting the networking plugin and associated driver code would be easier.

MattMead166, 2025-05-15:
Hello Terje, 

I will message Chris again and either confirm whether or not he is sending an updated SLB to you or have him reach out to you directly. I will also let Chris know to try a Web Builder build on his end as well as have some developers test also. 

I will give that a shot and see what I can do. I made some changes initially and after many fails I finally got a successful build, but unfortunately it seems that probe_state and toolsetter_state, even though I did everything I thought correctly  to add them, still don't work. They still only print "ok" when issuing (print, #<_probe_state>) or (print, #<_toolsetter_state>) and NOT [MESSAGE: 0 or 1]. 

I was looking in the STM32F4xx repo and I saw that someone else posted the same issue I am having back in February of '24. https://github.com/grblHAL/STM32F4xx/issues/161 and was wondering if that might have something to do with it also? In there you mention editing this line in report.c:

[image]

When I check my /core/report.c vs STM32F4xx/grbl/report.c vs sienciHAL/grbl/report.c (with sienciHAL working as expected) I see the following:

[image]

Would changing that line in my STM32F4xx/grbl/report.c yield similar results as the OP? Or do you think it is not worth trying?

terjeio, 2025-05-16:
> They still only print "ok" when issuing (print, #<_probe_state>) or (print, #<_toolsetter_state>) and NOT [MESSAGE: 0 or 1].

Support for _print_ was likely added after the current Sienci version - try _debug_ instead (remember to [enable it in settings](https://github.com/grblHAL/core/wiki/Expressions-and-flow-control#comments-with-parameter-substitution)).

> Would changing that line in my STM32F4xx/grbl/report.c yield similar results as the OP? Or do you think it is not worth trying?

The change was done to allow connecting via network/UART when the USB cable is _not_ plugged in or a client was not connected. Previous to that change sending data to the USB stack when no client was connected sometimes blocked communications (when alarms or other messages were output during initial startup?).
Are you back to the Web Builder version working with the USB cable plugged in?

MattMead166, 2025-05-16:
Hello Terje, 

Thank you for that update. I don't remember seeing anything specific regarding to "print" but I do remember seeing and adding "debug" in a few sections so I will reload that FW I built and try the command (debug, #<_probe_state>) and (debug, #<_toolsetter_state>) and see if that works. 

I tried to build a Web Builder version today after your update to 20250514 and am getting the errors below. I tracked it down to being "PWM2" when I add that as SPINDLE2, SPINDLE3, or SPINDLE4. If I do "ALL" or "MODVFD" as SPINDLE1 it builds successfully. Is that a bug or expected? Do I need to have PWM listed as SPINDLE1 for PWM2 to work?

I'm not on any specific version currently (Web Builder or VS Code or sienciHAL) as none of them work everything successfully. I have been testing and building versions multiple different ways trying to nail down a version that works correctly. I'm hoping the tip about "debug" instead of print will work and then I would be good again using my edited sienciHAL. All of my Web Builder versions "work" correctly after flashing. Even maybe once after power cycling the SLB but then it always reverts to the Ethernet not working. 

I only asked about adding/editing that line of code mentioned above as I saw the error posted in the STM32F4xx repo was the exact same issue I am describing just as of last year and not as of recent. 

Web Builder error LOG when adding PWM2 as SPINDLE2/3/4:

Command failed: pio run -etest -cjob.ini
motors/trinamic.c:563:17: warning: 'get_driver_enable' defined but not used [-Wunused-function]
static uint32_t get_driver_enable (setting_id_t setting)
^~~~~~~~~~~~~~~~~
motors/trinamic.c:550:22: warning: 'set_driver_enable' defined but not used [-Wunused-function]
static status_code_t set_driver_enable (setting_id_t id, uint_fast16_t value)
^~~~~~~~~~~~~~~~~
plugins/eventout.c: In function 'event_settings_restore':
plugins/eventout.c:309:38: warning: array subscript is above array bounds [-Warray-bounds]
plugin_settings.event[idx].trigger = Event_Ignore;
~~~~~~~~~~~~~~~~~~~~~^~~~~
In file included from networking/wiznet/enet.c:55:0:
networking/wiznet/port/w5x00_lwip.h:110:17: warning: 'ethernet_frame_crc' declared 'static' but never defined [-Wunused-function]
static uint32_t ethernet_frame_crc(const uint8_t *data, int length);
^~~~~~~~~~~~~~~~~~
networking/wiznet/port/w5x00_lwip.c:167:17: warning: 'ethernet_frame_crc' defined but not used [-Wunused-function]
static uint32_t ethernet_frame_crc(const uint8_t *data, int length)
^~~~~~~~~~~~~~~~~~
Src/driver_spindles.c: In function 'spindle1Config':
Src/driver_spindles.c:437:34: error: 'spindleSetStateVariable' undeclared (first use in this function); did you mean 'spindle1SetStateVariable'?
spindle->set_state = spindleSetStateVariable;
^~~~~~~~~~~~~~~~~~~~~~~
spindle1SetStateVariable
Src/driver_spindles.c:437:34: note: each undeclared identifier is reported only once for each function it appears in
*** [.pio/build/test/src/driver_spindles.o] Error 1
========================== [FAILED] Took 4.52 seconds ==========================

terjeio, 2025-05-18:
> I tried to build a Web Builder version today after your update to 20250514 and am getting the errors below. 

I have just commited a fix for this.

Can you ping the controller when you cannot connect?
Do you have a different machine you can try with? By different I mean a different make/model.

> I only asked about adding/editing that line of code mentioned above as I saw the error posted in the STM32F4xx repo was the exact same issue I am describing just as of last year and not as of recent.

Well, you can add it - it should fix any hangs you may experience with the Sienci fork. It should not cause hangs with the upstream code since it is there (elsewhere in the code now) to prevent hangs if the USB cable is not connected or no client is connected via USB.

MattMead166, 2025-05-18:
Unfortunately still getting errors building after you posted fix. 

Command failed: pio run -etest -cjob.ini
motors/trinamic.c:563:17: warning: 'get_driver_enable' defined but not used [-Wunused-function]
static uint32_t get_driver_enable (setting_id_t setting)
^~~~~~~~~~~~~~~~~
motors/trinamic.c:550:22: warning: 'set_driver_enable' defined but not used [-Wunused-function]
static status_code_t set_driver_enable (setting_id_t id, uint_fast16_t value)
^~~~~~~~~~~~~~~~~
plugins/eventout.c: In function 'event_settings_restore':
plugins/eventout.c:309:38: warning: array subscript is above array bounds [-Warray-bounds]
plugin_settings.event[idx].trigger = Event_Ignore;
~~~~~~~~~~~~~~~~~~~~~^~~~~
networking/wiznet/port/w5x00_lwip.c:167:17: warning: 'ethernet_frame_crc' defined but not used [-Wunused-function]
static uint32_t ethernet_frame_crc(const uint8_t *data, int length)
^~~~~~~~~~~~~~~~~~
In file included from networking/wiznet/enet.c:55:0:
networking/wiznet/port/w5x00_lwip.h:110:17: warning: 'ethernet_frame_crc' declared 'static' but never defined [-Wunused-function]
static uint32_t ethernet_frame_crc(const uint8_t *data, int length);
^~~~~~~~~~~~~~~~~~
Src/ioports_analog.c: In function 'analog_in_state':
Src/ioports_analog.c:201:114: error: 'input_signal_t {aka struct <anonymous>}' has no member named 'channel'
return input->id < analog.in.n_ports ? (float)adc_read(aux_in_analog[input->id].adc, aux_in_analog[input->id].channel) : -1.0f;
^
Src/ioports_analog.c: In function 'wait_on_input':
Src/ioports_analog.c:206:92: error: 'input_signal_t {aka struct <anonymous>}' has no member named 'channel'
return port < analog.in.n_ports ? adc_read(aux_in_analog[port].adc, aux_in_analog[port].channel) : -1;
^
Src/ioports_analog.c: In function 'ioports_init_analog':
Src/ioports_analog.c:333:76: error: 'input_signal_t {aka struct <anonymous>}' has no member named 'channel'
adc_config.Channel = aux_inputs->pins.inputs[i].channel = adc_map[j].ch;
^
Src/ioports_analog.c: In function 'analog_in_state':
Src/ioports_analog.c:202:1: warning: control reaches end of non-void function [-Wreturn-type]
}
^
Src/ioports_analog.c: In function 'wait_on_input':
Src/ioports_analog.c:207:1: warning: control reaches end of non-void function [-Wreturn-type]
}
^
*** [.pio/build/test/src/ioports_analog.o] Error 1
========================== [FAILED] Took 4.50 seconds ==========================

terjeio, 2025-05-19:
> Unfortunately I am still getting errors building after you posted fix.

Oops - missed a file, fixed now.

MattMead166, 2025-05-21:
Hello Terje, 

So I was able to make a build 2 days ago after seeing your message. Good news is everything seems to be working again. I have power cycled many times over the past few days and my Ethernet connection has connected and remained reliable without the use of a USB cord first. So far so good. I am not going to change anything for now as this FW seems to be working as expected. Fingers crossed everything remains stable. Though I did notice after testing that my computer has another update available but I have not done it as that's when I had issues arise last time was after updating. 

I want to say thank you so much for all your help and dedication to this issue. I appreciate it very much and all the work you do for the community. It took some time for sure but you never gave up and I really appreciate that. 

The only thing I did notice is toolsetter_state doesnt quite seem to be working properly in my build. It seems to be returning weird values. I am going to go through all my wiring and make sure nothing got disconnected or messed up. Again I just want to thank you for all that you have done and continue to do.

terjeio, 2025-05-21:
> Good news is everything seems to be working again.

That's great. I still have occasional issues with my main development machine.

> The only thing I did notice is toolsetter_state doesnt quite seem to be working properly in my build.

See footnote 3 at the bottom of the [expressions and flow control](https://github.com/grblHAL/core/wiki/Expressions-and-flow-control) Wiki page. One reason for for this is that the SLB combines the probe and toolsetter inputs by ANDing or ORing them depending on the probe direction. This is a creative workaround for a limitation in the core - it can only handle one probe input at any given moment.

To use more than one probe, and keep their states unambiguous, one had to use a single input and connect the probe to be used manually, or have a relay (or electronic switch) doing it - either by a [M-code or automatically](https://github.com/grblHAL/Templates/blob/5385b30376d72ecdaceb3b3b1fff0e720b6a6cd0/my_plugin/Probe_select/my_plugin.c#L17-L24). Or add a second probe input and be creative.
[A few commits ago](https://github.com/grblHAL/core/blob/master/changelog.md#202501514) I added HAL call for probe selection, new options to `$346`, and an inbuilt `G65` macro available if more than one probe is available. This means that a driver (or a plugin) can now register up to 3 probes/probe inputs. I have already made a [new plugin](https://github.com/grblHAL/Plugins_misc/#probe-relays) for relay-based selection but have not yet added generic driver support for additional probe inputs.
So what to do with the SLB approach that combines the two inputs? Drop that, add generic driver support (I'll do that anyway) and force explicit probe selection? This will make the returned `_probe_state` and `_toolsetter_state` values unambigous, but risks creating issues with established workflows...

closed 2025-06-02 by terjeio