grblHAL/core #740: Laser in continuous fire after USB connection problem

Stephan-S-1983, 2025-05-04:
Hello,
I am currently testing a DLC32 MAX in conjunction with a 100W CO2 laser.
During the tests, a connection error occurred once on the USB interface. The laser remained at the last current position, but the laser continued to fire continuously. Is there already a possibility to activate a monitoring, which deactivates PWM and laser ON signal in case of an error? If not, this would be an important function to implement for safety reasons

Best regards,
Stephan

terjeio, 2025-05-06:
> Is there already a possibility to activate a monitoring, which deactivates PWM and laser ON signal in case of an error?

Maybe - a plugin could be made that moniors for loss of status requests from the sender. And perhaps monitor for loss of the DTR signal or a network connection.

hanke-cnc, 2025-05-07:
> > Is there already a possibility to activate a monitoring, which deactivates PWM and laser ON signal in case of an error?
> 
> Maybe - a plugin could be made that moniors for loss of status requests from the sender. And perhaps monitor for loss of the DTR signal or a network connection.

"I have a brilliant idea that might be able to help him: what if the laser turns off if no '?' is received from the host computer (or master device) within one second?
Additionally, Terjeio, your latest F4 code doesn't seem to be working. At least, it's not working on my F407. I only tested the standard serial port (PA9, PA10), and it showed no reaction at all. The source code was unmodified. I have a version of the source code on my computer that I downloaded in March, and that one works correctly. So, I suspect the issue isn't on my side.
Since I've moved on to the RP2350, I haven't raised this issue in the F4 section/forum."