grblHAL/core #749: X1 and Z motors locking rather than jogging

ChancellorVerdery, 2025-05-23:
Board: BTT Octopus Max EZ
Drivers: EZ2209
Motors: Nema 17

I built a dual motor XY single motor Z machine. I have X_GANGED, Y_GANGED, X_AUTO_SQUARE,  and Y_AUTO_SQUARE defined in my_machine.h

I have TRINAMIC_ENABLE set to 2209 and TRINAMIC_R_SENSE set to 110

I also have the board defined, BOARD_BTT_OCTOPUS_MAX

For the Board map I am using btt_octopus_max_map.h

When G-Code is sent to the Y axis, there are no issues. When sent to the X axis, only the X motor jogs and the X1 locks. When sent to the Z axis, the Z motor locks and doesn't jog. 

What I have tried:
Swapping motors
Changing the pinout
Swapping to a different EZ2209 driver
Reversing motors
Using the Web Builder generated firmware

I'm lost as to what what I could be doing wrong, so it seems to be an issue with the firmware.

terjeio, 2025-05-23:
I've done a quick test with the F7xx driver which the [H7xx](https://github.com/dresco/STM32H7xx) is based on and that works (with a different board map).

The pin map for the Octopus seems to be ok, @dresco do you have any ideas?

dresco, 2025-05-24:
Hi, I've not got a Octopus Max board to test with, but do have the BTT SKR3 and Scylla with Trinamic drivers.

@ChancellorVerdery - a few initial thoughts;
Does the `$pins` output tie up with whats expected from the board map & schematic?
Are you able to temporarily configure as 5-axis, to test each motor output individually? 
Also worth checking the `M122` output to ensure the TMC drivers are being detected/configured as expected?

ChancellorVerdery, 2025-05-26:
Sorry for the late reply I've been very busy this weekend.

I just got all axes working. The issue laid in the order that the steppers were plugged into the board in addition to which ports they are plugged into. 

If the Z axis stepper was plugged into a port preceding the X1 port then the Z stepper wouldn't work. For example, if Z was plugged into motor port 3 and X1 was plugged into motor port 4, then neither would work. I tested moving both Z and X1 to different motor ports and I got X1 functionality; however, Z would only move down, no matter if the G-Code told it to go up or down. Finally, I switched the ports for Z and X1 so that Z was in motor port 10 and X1 was in motor port 7. That got all axes working properly. It seems that I may have a faulty board or some logic doesn't work properly with my board. 

Let me know if you want any more information to figure out if it is a board issue or firmware issue.

closed 2025-05-26 by ChancellorVerdery