grblHAL/core #753: is there a way to debug the MODVFD plugin?

pmcquay, 2025-06-02:
I've gotten my VFD working with the MODVFD plugin, but I've got issues with error 14 happening randomly whenever there isnt a job running, as well as the machine not waiting for the spindle to be at speed before continuing on with the job. More details below.

1. it seems random, but when there isnt a job running, the machine will sometimes just fire an error 14. I haven't been able to deduce a pattern. It can happen when the spindle is running (say, during spindle warmup) or not running, it can happen on estop clear (My estop removes VFD power but not control board power, so I would expect some strangeness from the modbus network. AFAIK this VFD does not have STO). I haven't had it happen during a job yet, but I cannot say that it will not happen.
2. if the spindle is not running, and I start a job, it will start and wait correctly. If there is a job in progress that contains a speed change, it does not seem to wait for the speed to change. Similarly, if there is no job running, but the spindle is on, and a job is started that has a significantly different speed, the machine will not wait. I have only seen this last behaviour one time, so I cant say if it happens all the time.

The VFD is a Fuling BD-612 5.5kw, I am running a PicoCNC board with one spindle configured as MODVFD, and I have the speed tolerance set to 10%, and the timeout set to 20 seconds. I believe I have the latest firmware from the web builder? I am not 100% certain of that. The connection is a simple unshielded twisted pair that is about 50cm long. Everything else in the enclosure is double shielded and properly grounded. Its running 9600 baud 8N1. Other config is listed here: https://www.reddit.com/r/hobbycnc/comments/1kt4u3e/grblhal_modbus_settings_for_fulingfolinn/

Is there any way that I can see the number of message retries on the modbus line? or see the data that is sent back and forth without a logic analyzer? Are there other things I can check that would narrow down the problem?

terjeio, 2025-06-05:
> ... the machine not waiting for the spindle to be at speed before continuing on with the job.

What is your $340 setting? If 0 spindle at speed check is disabled and $394 spindle on delay is observed instead.

> If there is a job in progress that contains a speed change, it does not seem to wait for the speed to change.

There is no delay implemented for RPM changes - you will have to add G4 pauses to your post for that.

> Is there any way that I can see the number of message retries on the modbus line?

I have added a new $command, `$MODBUSSTATS`, in the latest commit that outputs information about number of commands sent, retries etc.:

`[MSG:Info: TX: 16811, retries: 0, timeouts: 0, RX exceptions: 0, CRC errors: 0]`

> or see the data that is sent back and forth without a logic analyzer?

Easiest way to monitor the traffic would be to hook up the RX line of a  USB <> UART breakout to the modbus RX and/or the TX lines. Alternatively I have to echo transferred data/statuses to a debug UART port, possible to do but will take some time to implement and test. For now I have debugged the communication with a logic analyzer and the MCU debugger. What is tricky to do though is to introduce faults in the communication to check that they are correctly handled...

patrickmcquay, 2025-06-06:
Ahh, interesting. $340 is 10, so I would assume that means a 10% tolerance positive and negative.

Seems odd to wait on startup but not speed changes, I'm curious as to the rationale behind that decision? Regardless, if thats expected then fair enough.

Is that command available in the web builder?

patrickmcquay, 2025-06-06:
Fair enough. I suppose I come from the 3d printer world, where the prevailing notion is that the machine should just know how to run itself, as much as its possible to do so. CAM for subtractive machines is somewhat more vague that way, and I'm still learning. I can think of at least one reason I wouldnt want to wait on spindle speed changes with a lathe as well, that would be harder to implement inside the machine, constant surface speed for varying diameter cuts. It really depends on what information the machine has about whats actually happening in that case.

Ill grab the latest web builder version soon and see if a. It fixes the error 14 on unlock and b. I can see anything in the new debug command.