grblHAL/core #766: Homing question/request

Aepx, 2025-06-27:
I know you can home and set that position to 0/0.
Is there a way to home and set that position to other coordinates?

I want 0/0 to be lower left but I home to upper left.
My other software I use on another machine let's you set the home position to any coordinates you want but I don't see a way in grblhal.
Is this possible now or can it be added?

Aepx, 2025-07-02:
Working in the negative is definitely possible and not a problem until you want to use absolute coordinates. Then you have to remember it's negative. That's why other software allows for setting the home to whatever coordinates you want which keeps the absolute in the positive.
There are easy workarounds, like the link you posted, but just being able to enter it in the config would be more elegant.

terjeio, 2025-07-02:
> Is there a way to home and set that position to other coordinates?

Yes, a plugin can be written that expands on the current functionality.
Internally a work envelope is kept that holds the min and max values of the machine coordinate system. [This is set](https://github.com/grblHAL/core/blob/df55d7cb916646a12e9b35eb33cabd2efcbf805b/machine_limits.c#L114-L140) prior to firing the [on_homing_completed](https://github.com/grblHAL/core/blob/df55d7cb916646a12e9b35eb33cabd2efcbf805b/motion_control.c#L970-L974) event. If subscribed to by a plugin it can then set the work envelope to whatever it sees fit.

In addition the functions for handling limits checks can also be overridden by a plugin since they are called [via function pointers](https://github.com/grblHAL/core/blob/df55d7cb916646a12e9b35eb33cabd2efcbf805b/machine_limits.c#L851-L853) initialized on boot. If overridden all kinds of work envelopes can be implemented, circular, oval, ... These do not have to use the internal work envelope variables. If the internal variables are used and the work envelope is rectangular it is a fair chance that the provided functions will work as-is - though this needs to be verified.

terjeio, 2025-07-05:
Adding the option to the core will trigger a settings revision which I am not yet ready for. A plugin can add settings without affecting the core.

Aepx, 2025-08-01:
I think I found another reason to add this but maybe I'm just missing it. 
Is there an offset for homing?
What I mean is if the home switch is farther negative than zero, in other programs (even Marlin for printers) you can tell the software to offset by the amount of the negative value. 
Is this already available somewhere?

terjeio, 2025-08-01:
$27 is the homing pull off distance, not sure if this is what you want.

Aepx, 2026-05-09:
Has this been added to the latest release?