grblHAL/core #775: How to get bootloader mode on on mks tinybee.

Shiraz2110, 2025-07-14:
I once uploded marlin on mks tinybee. Now i need to update the firmwire.
But everytime i want o upload the firmwire it shows "can not connect to esp 32" 
I wanted to turn on bootloader mode. But i can not find any physical button. There is one white button but pressing it doesn't work. 
Is there any way? How to turn bootloader mode on?

whmountains, 2025-07-15:
After checking the schematic for the tinybee, that white button is just the reset button.

You shouldn't need to put it into "bootloader mode" in the same way as the RP2040 or similar.  The USB-UART converter on the board should handle resetting the chip automatically.

My first guess would be that the baud rate is incorrect.  The [platformio.ini](https://github.com/grblHAL/ESP32/blob/master/platformio.ini#L42) file for grlbHAL/ESP32 seems to set the baud rate at the standard `115200`.  I don't know if it's the official Marlin but I found [this repo](https://github.com/zhenweiws/Marlin-esp32/blob/Yboard/ini/esp32.ini#L21) which sets it at a blazing `500000`.

If that doesn't work, have a look at this issue and try the troubleshooting steps suggested there: https://github.com/bdring/FluidNC/issues/491.  Ultimately the problem was not fixed in that case, but they tested a LOT of things along the way, and maybe one of those lead you to the solution.

terjeio, 2025-07-16:
> My first guess would be that the baud rate is incorrect. 

The bootloader auto-detects the baud rate.