grblHAL/core #810: how to find what pins the web builder is using

connerkidwell, 2025-09-20:
I am using a BTT Octopus max ez board on a homemade diode laser in a core xy kinematic system. I have used the web builder to generate the code for the board. I have everything working correctly, auto squared z axis, xy motion, homing and I can even fire the laser and change power. Somewhere I came across info and decided to use the fan 4 port for my pwm and ground to the laser. The power for the laser comes straight from my power supply. I have noticed that if I move the laser manually, I get enough back emf from the motors to fire the laser. AI suggested that I add a pulldown diode to the pwm and ground to prevent noise from triggering it to fire, at some point it also suggested that the fan4 port wasn't a good port to use. I am unsure if I am using the incorrect port for pwm and ground and that is causing my issues or if I have some other problem. I did try adding the pulldown resistor and it didn't seem to make a difference.

connerkidwell, 2025-09-20:
thank you, that confirms that PA1 is the correct pwm pin and that is in fact the fan 4 port. I am also running the ground to the fan 4 ground, does this seem correct?

terjeio, 2025-09-21:
Looks correct to me, P11 output is a buffered 5V signal (74HCT365). You should be careful about rotating the motors too fast manually, the back emf may kill your drivers - or even the board. Some drivers may have back emf protection, check their specs.

When checking the schematic I see that AI did not...

connerkidwell, 2025-09-21:
Ok, thank you for your help! I have the ez2209 drivers and from what I see they don't have back emf protection so I will just have to avoid moving it by hand. When I turn my power switch on the laser fires for like 5-10sec presumably until the board boots up. Do you have any ideas how I could prevent that from happening. I have my power ran straight from the power supply to the laser board. I would imagine I could just put a physical switch to enable or disable the power to the laser. But I was wondering if there was a more elegant solution with like a relay or something maybe from the cycle start output to enable the laser? I was planning to add cycle start and feed hold buttons I just don't have the wired up. Or would it be better to have a way to have it wired to the e-stop and just release the e-stop after turning the machine on because the board would be booted up by then? Thank you for being so helpful!

terjeio, 2025-09-21:
> Do you have any ideas how I could prevent that from happening?

A pulldown resistor on the PWM output (or enable output if you use that) from the MCU (at the 74HCT365) could do the trick. This since MCU pins are typically high-Z during boot. It would require careful soldering...

connerkidwell, 2025-09-21:
Thanks for all the help and advice, I will probably just live with it. Careful soldering isn't something I'm super good at 😂

closed 2025-09-21 by connerkidwell