grblHAL/core #812: Ov: report is missing while in Run mode

karoria, 2025-09-21:
Hi @terjeio 
From my last few compilations, I am stuck with a strange situation where in Run mode, report doesn't include "Ov:" field. In idle mode Ov: is included in report. Kindly guide where I am wrong. My $10 setting is 1023 so it might not be an issue. My build details are as under:

[VER:1.1f.20250313:valay]
[OPT:VNMZHS+,100,1024,5,8]
[AXS:5:XYZAB]
[NEWOPT:ENUMS,RT+,HOME,ES,REBOOT,EXPR,SED,ETH,FTP,FS,DISPLAY,SD]
[FIRMWARE:grblHAL]
[SIGNALS:HSDEFP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:484K]
[DRIVER:iMXRT1062]
[DRIVER VERSION:250228]
[DRIVER OPTIONS:USB.2]
[AUX IO:4,7,0,0]
[IP:192.168.1.77]
[NETCON:Websocket]
[PLUGIN:FS stream v1.00]
[PLUGIN:FS macro plugin v0.17]
[PLUGIN:WebUI v0.27]
[PLUGIN:Homing pulloff v0.01]
[PLUGIN:Persistent tool v0.01]
[PLUGIN:Display v1.0.0 - (SSD1306 I2C not connected)]
[PLUGIN:SDCARD v1.23]
ok

Waiting for your answer.
Thanks.
Ravi

terjeio, 2025-09-21:
I have changed the override reporting in later versions so at least it works now. So it must be due to an earlier regression?

karoria, 2025-09-21:
Ok. Compiled latest build. It works!

closed 2025-09-21 by karoria